SlimeVR/SlimeVR-Server #12: Implement reading floor level from the driver

Eirenliel, 2021-07-13:
Floor level is not always 0, this can screw up many functions that rely on it, including auto skeleton calibration. Need to find it in the driver, record it, and use as a base.

Louka3000, 2022-04-16:
As @kitlith  said,
> for floor level: if you pass an argument to the feeder app (`--universe=standing`), y=0 in that space is floor height. This would transform the problem from "how to get floor height" to "how to use standing universe measurements in slimevr server/preferring the feeder app over the driver"

We should use standing universe for the hmd and controllers. This gives us the positions relative to floor level.
Then, if we need to send back the positions in raw universe, we can easily offset the positions by the difference between the driver HMD and the feeder HMD

Eirenliel, 2022-04-16:
or we make driver/feeder send the difference between standing and raw to the server, and server applying this transformation to all data it recieves/sends. unless we can make driver fully work in standing universe too

Louka3000, 2022-04-16:
So we either
1. Make the driver work in standing universe
2. Or offset all our data with help of the feeder app.

1 is better IMO but it might not be possible. Also need to make sure this doesn’t break Quest standalone support!!

kitlith, 2022-04-16:
> or we make driver/feeder send the difference between standing and raw to the server, and server applying this transformation to all data it recieves/sends. unless we can make driver fully work in standing universe too

there is a file that stores the difference between raw and standing universe, if you want to handle this in the driver. It's `C:\Program Files (x86)\Steam\config\chaperone_info.vrchap` on my machine. I don't know what the reliable way to get that path is (i.e. steam installed in custom location, on a another drive, idk) or how to detect if it refreshes, but it does exist.

kitlith, 2022-05-30:
Had a productive conversation with @Louka3000 and @Eirenliel in discord today, where we decided exactly where and how this should be implemented in the slimevr ecosystem.

Specifically, we decided this is a driver issue, and that the driver should be sending/receiving positions in standing universe by reading the json file I mentioned in my previous post, and using it to translate the raw universe positions. (I'll look into transferring this issue to the correct repository once I finish writing this post)

An initial/proof of concept implementation shouldn't be too difficult. The main concern is making sure that we find `chaperone_info.vrchap`, no matter what system we're running on. Likely, we should rely on a registry key, or some way of querying steam itself, to point us in the correct direction. `Computer\HKEY_LOCAL_MACHINE\SOFTWARE\WOW6432Node\Valve\Steam\InstallPath` looks promising, granted the likely difference between 64 bit machines and 32 bit machines. I welcome someone who is more familiar with this type of problem to chip in.

I'll be available to help anyone who picks this issue up.