SlimeVR/SlimeVR-Server #13: Multi-step tracker calibration support

Eirenliel, 2021-07-13:
Need to add calibration code and GUI for trackers that require calibration. It should be multi-step controlled by the server. For example, user clicks on "Calibrate gyroscope", server ask the user to put trackers down, and then records some data for gyroscope offset. Same for accelerometer and magnetometer calibration. Different calibrations should depend on the board and sensor used. owoTrack app should say it doesn't need calibration from the server, just move your phone in the 8 pattern, etc.

ButterscotchV, 2024-09-17:
To update this issue, calibration has largely been implemented purely in firmware, having a GUI implementation for 6-side calibration and such could be beneficial.

ImUrX, 2024-09-18:
wdym? rn we have a tutorial, you mean a way to trigger it from server?

ButterscotchV, 2024-09-18:
> wdym? rn we have a tutorial, you mean a way to trigger it from server?

Trigger and guide the 6-side calibration from the server