SlimeVR/SlimeVR-Server #7: Leg twist compenstation

Eirenliel, 2021-07-13:
Some people's legs twist when they bend or twist them, like on the pictures below:

[image]

[image]

We need to add a coefficient that can be configured when the person sits, that will compensate for the twist. The coefficient should untwist the leg based on the pitch of it. The compensation should be applied to the base rotation of the tracker.

ButterscotchV, 2022-03-24:
Just coming in to document what I had previously toyed around with for this. I messed around with the skeleton data and offset the lower leg tracker rotation by a manually calibrated amount, multiplied by the bending of the knee, doing so I managed to somewhat consistently reduce the false rotation which resulted in significantly less foot sliding.

For the manually calibrated values, each leg had to have its own calibrated value, they were significantly different. The issues left to resolve for this are how it will be calibrated, whether it would need to be calibrated every time the trackers are put on, and how to compensate for the rotation of the upper leg.

Eirenliel, 2022-04-11:
[image] [Leg twist compensation](https://trello.com/c/d383fl6w/39-leg-twist-compensation)

loucass003, 2025-09-24:
Cleaning up old issues. Do we still want this? @Eirenliel @ButterscotchV