SlimeVR/SlimeVR-Server #8: Ineractive skeleton settings and calibration setup

Eirenliel, 2021-07-13:
There can be a lot of things to improve calibration (reset) and skeleton settings that will be more precise than pressing reset while standing still, or asking users to input their measurements or align trackers to their body. Here will be a list of ideas. We will need a framework to implement them in the future, and add more.

* Bend leg in the knee and raise it up to configure forward axis. The yaw 0 will be on the plane that goes through the angle made between the start and the end of the bend, parallel to the floor. This will eliminate the need to put trackers on exactly facing one of the cardinal sides.
* Bend forward to configure the yaw 0 for chest and waist trackers, similar to the leg.
* Move foot up and down to configure the yaw 0 for the foot.

Eirenliel, 2021-07-13:
Hiratora uses similar approach to aligning yaw axis with squatting, can be used too for both legs and body at the same time

[image]

Eirenliel, 2022-04-11:
Superceed by #148

Eirenliel, 2022-04-11:
[image] [Mounting calibration](https://trello.com/c/35OHV2tW/9-mounting-calibration)

Louka3000, 2022-04-30:
Working branch: https://github.com/Louka3000/SlimeVR-Server/tree/mounting-calibration

Louka3000, 2022-10-05:
PR for this: https://github.com/SlimeVR/SlimeVR-Server/pull/251/

closed 2022-11-22 by Eirenliel