SlimeVR/SlimeVR-Server #25: Natural Locomotion support

Eirenliel, 2021-08-17:
To support [Natural Locomotion](https://store.steampowered.com/app/798810/Natural_Locomotion/) SlimeVR need to report speed (or acceleration) values in the driver. This requires support from firmware, software, and driver. Creating issue here for this feature, but it needs to be implemented everywhere.

Louka3000, 2022-10-06:
What's the status on this?
NaLo no longer crashes... I think all we need to do is forward accel data to SteamVR now, right?
Conveniently, Stermere already did most of the job ;)

ruti75, 2025-10-07:
Hello,
Is there an update on this? I've been trying to write my own code (only using the ankle trackers for simplicoty) and I can share it but I'm a total noob amd was hoping someone did a better job than me.