SlimeVR/SlimeVR-Server #26: Implement ways to reset tracker without having to open Server window.

Torikari, 2021-08-17:
Either through tapping the tracker, controller button press and gesture, or any other potential idea.

This will be a huge QOL upgrade to people who moves a lot, having the need to reset tracker in multiple succession for experimenting tracker adjustments, or unlucky enough to just have a lot of drift happening to them, worth to be in the issue here as a constant reminder.

Eirenliel, 2021-08-17:
Is it fine if it's Fast Reset not full reset? So there is less chance to screw up alignment by body not being pefrectly straight when doing it?

Torikari, 2021-08-17:
> Is it fine if it's Fast Reset not full reset? So there is less chance to screw up alignment by body not being pefrectly straight when doing it?

Preferably Fast Reset actually, it's been working okay for me for the past week, but it's probably good to always have a choice.

Louka3000, 2022-04-11:
[image] [Reset without SlimeVR window or OVRAS keybind.](https://trello.com/c/PPRMLj2d/30-reset-without-slimevr-window-or-ovras-keybind)

ButterscotchV, 2022-12-16:
Please re-open if this is not fully fulfilled, but I believe #328 has fulfilled this issue and will be further expanded beyond this issue with #375.

Eirenliel, 2022-12-16:
I would love it to have integration with steam controller binding via feeder or overlay though...

ButterscotchV, 2023-12-14:
> I would love it to have integration with steam controller binding via feeder or overlay though... But maybe it's overkill at this point, we have a bunch of options now. It's technically implemented, yes.

This is implemented in the feeder now, closing this issue.

closed 2023-12-14 by ButterscotchV