SlimeVR/SlimeVR-Server #33: Packet ID conflict with owoTrack Android application

abb128, 2021-08-22:
Pardon my poor documentation, but packet IDs 5 and 6 are actually already in use by the owoTrack Android app.

This code appears to be using these packet IDs for other purposes: https://github.com/SlimeVR/SlimeVR-Server/blob/5c6d02de30d646bf095fe190a1bc2245e6de7a17/src/main/java/io/eiren/vr/trackers/TrackersUDPServer.java#L235L249

These aren't implemented in the owoTrack server, but the Android application sends these packets anyway. This might be causing some issues.

Packet 5 tells whether or not the magnetometer is enabled with a single char 'y' or 'n': https://github.com/abb128/owoTrackVRSyncMobile/blob/11f6718337aafce2db8daeb62076bb0e9129cee0/app/src/main/java/org/owoTrackVRSync/UDPGyroProviderClient.java#L254L269

Packet 6 is sent when the power button is pressed and the device screen is turned on (the method name is recenter_yaw, but that was just a suggestion): https://github.com/abb128/owoTrackVRSyncMobile/blob/11f6718337aafce2db8daeb62076bb0e9129cee0/app/src/main/java/org/owoTrackVRSync/UDPGyroProviderClient.java#L271L278

adigyran, 2021-08-22:
Can we try to make SlimeVR version of owo App?
:)

Eirenliel, 2021-08-22:
> Can we try to make SlimeVR version of owo App?

why.

> Pardon my poor documentation, but packet IDs 5 and 6 are actually already in use by the owoTrack Android app.

They were different or unused when I started :D 

Fixed in https://github.com/SlimeVR/SlimeVR-Server/commit/9383be678cf53175b677c64f429a86ff7bbe396a

closed 2021-08-22 by Eirenliel