SlimeVR/SlimeVR-Server #36: Autoupdater

adigyran, 2021-08-23:
Implement some sort of auto updater to server, for easy distribution

Eirenliel, 2022-04-11:
[image] [Server autoupdate](https://trello.com/c/8iq01n1O/33-server-autoupdate)

loucass003, 2022-07-10:
What are the main ideas for this?

Should also only the server updates itself. Or should it also update the new GUI with it ?
Maybe both could have their own self-update cycle separatly ?

Eirenliel, 2022-07-10:
it should be one thing along with the driver

loucass003, 2022-07-10:
So Driver Server Gui ? Or only Driver Server ?

Eirenliel, 2022-07-10:
everything. also overlay.

Eirenliel, 2022-07-10:
and the feeder if it's still separate from the overlay when it's ready.

loucass003, 2022-07-10:
What is you reasoning behind this ? Because i think having separate updates could be pretty usefull. Like being able to update the gui without he server or the overlay without the server

Eirenliel, 2022-07-10:
Eh, it's basic usability? Don't forget that end-user is not a DIY person who wants everything configured and run multiple versions of everything. One update process should update everything there is to update.

TheDevMinerTV, 2022-07-10:
Also remember that would have to insure that the protocol between all the applications is backwards compatible, otherwise this would be a pain in the butt to use.

Eirenliel, 2022-07-10:
Why would it be a problem, all things should be updated simultaneously with this.

TheButlah, 2023-02-12:
I wish to tackle this - I'll be starting by building a cli tool to install the overlay. The existing windows installer can call that cli tool. I'll be modeling it after [rustup](https://rust-lang.github.io/rustup/concepts/index.html#terminology), where it manages a set of components with versions, and we have channels (like stable) that dictate the versions for each component.

The existing MSI installer will call into the CLI, so we retain the nice UX of the installer while removing the need for so much windows-specific msi scripting.

Sacred-Shadow, 2026-03-22:
Is Issue 2026 still relevant? Now that we're “just” about to release on Steam?