SlimeVR/SlimeVR-Server #42: Constant SteamVR recalibration prompts

oroshikirin11, 2021-08-31:
When using all available trackers in the server, SteamVR will prompt you to recalibrate your playspave every time you restart the server. If you toggle the trackers down, the old playspace still loads fine.
There was some debate wether it is cause by base station tracked headsets  but no confirmation yet.

Louka3000, 2021-09-30:
Just to be clear, this only happens when you use 7 trackers: “Waist + Chest + Knees + Feet/Ankles”
Using only 3,4 or 6 trackers works fine.
So current workaround would be to use “Waist + Chest + Feet” or “Waist + Knees + Feet”
Happens to me on Valve Index (base station tracked) btw.

Eirenliel, 2021-09-30:
Here's some hope new driver fixes it. Should be up next week 🤞