SlimeVR/SlimeVR-Server #47: It is hoped that a mode can use the same IP to transmit all nodes

salier, 2021-09-13:
I'm using a visual capture system. Because the data is transmitted by the same device, the IP is the same. I want to know how to distinguish different nodes.

https://user-images.githubusercontent.com/28691601/133083616-db7f672b-c924-407a-b907-05169ac0523c.mp4

The effect of the program is about this

Eirenliel, 2021-09-13:
One IP can register multiple trackers by sending multiple PACKET_SENSOR_INFO (15) packets, and then use PACKET_ROTATION_DATA (17) with different sensor ID to send data

salier, 2021-09-13:
Because I am not very good at code, I want to ask how to send bytes
For example, I now see that the code handshake of ESP should be 3 0 0 4 1 1 1?
Then 1 (rotation) 0 (coding) and then quaternion four numbers
I didn't find the problem of how to distinguish between ports and multi nodes

Eirenliel, 2021-09-13:
If you aren't very good at code, it would be hard for me to help you. Please ask in our Discord where others may help you instead, sorry

closed 2021-09-18 by Eirenliel