SlimeVR/SlimeVR-Server #50: Add BVH recording functionality

Eirenliel, 2021-09-19:
Need to add functionality to record [BVH files](https://en.wikipedia.org/wiki/Biovision_Hierarchy) with animations. This should have start and stop buttons/commands, and record all tracker animations in a new file.

Eirenliel, 2021-09-19:
We should reuse or refactor autobon recorder - PoseFrame. Probably separate classes for capturing and writing data, so we could reuse code for burst data writing of PoseFrame and streaming of BVH.

adigyran, 2021-09-21:
Livelink !

kitlith, 2022-01-17:
What is the status of this issue, now that #89 has been merged?

ButterscotchVanilla, 2022-01-17:
There's no GUI yet and I believe that BVH is still only *mostly* functional, will look into it later, just haven't had much time recently. I also want to refactor TransformNode a bit so that I can get rid of my messy wrapper for it and just re-use the class.

ButterscotchV, 2022-03-24:
To update this issue, BVH is pretty much fully working now, from here, most stuff would be improvements in functionality, configurability, etc.

closed 2022-03-24 by Eirenliel