SlimeVR/SlimeVR-Server #74: Create port input field on GUI to make it possible for multiple users on the same network to use Slimes

calliePepper, 2021-11-01:
All it needs is an input field, would be easy to build into the doc pages while setting up firmware to change the port field there

unlogisch04, 2022-03-29:
I think changing Port is not the way to go. You would also need the Firmware to change Port too. 
That's too much User Interaction and Complex for a User todo. For now maybe a dirty fix.

I got today to the same Issue in Chat. I came up with a different solution.
My first idea, but im not a java programmer:
1. Server Saves the MAC-Address of his trackers and per default only answers to them on Broadcasts. Config Option for still answer to all (Test / DIY)
2. Server Needs a method to Assign and Unassigned trackers. (Maybe like a Search for new Trackers)
3. Server needs to save that config data.
4. Assign trackers to a HMD?
5. Export + Import Trackerset + Userset Data (Tracker assignment + Body Portion)

Eirenliel, 2022-03-30:
I have a few ideas too, but I forgot them. I think port can be one of the options, and server can in theory handle it automatically via USB command too.

unlogisch04, 2022-03-30:
I think to change this stuff should not require a plug to USB. It should be easy a possible (user friendly). 
You could have a Control connection and a Data connection like ftp/http. 
Like:
main port = control connection
random port = data connection
For at the moment i do not see a reason to have that, or what benefit it would have.

I had a discussion in voice with @loucass003
The following Ideas came up:
1. Save the Data on the Server with the MAC-Address or some sort of Tracker ID
2. Save the Data on the Tracker. (Tracker group, Tracker assigned Position)

All this will add a "additional" Configuration step, of assign the trackers to a group and safe this.

ButterscotchV, 2023-01-16:
To note what's been done so far, there's been a quick workaround entry added to the config in #383 to allow manually setting the port. Further work is required for this issue, but now there's a workaround available.

Eirenliel, 2023-04-06:
Server-side implementation that doesn't require protocol would look like this:
- Server receives handshake multicast request
- If the server knows this tracker (MAC is in the config), the handshake is automatically accepted
- If user is on the screen "connect trackers", the handshake is automatically accepted
- Otherwise a popup created that says "New tracker found, add?"

PR #900 (2024-03-15): Allow multiple servers in the same network
Depends on https://github.com/SlimeVR/SolarXR-Protocol/pull/127

This adds the following features:
* A list of known devices that only works with mac addresses
* A new model version for config and migrate by adding all known trackers that are mac addresses to a set called `knownDevices`
* A way to forget connected devices (FIXME: for some reason it ends up disconnecting after 30 seconds have passed instead of immediately)
* A prompt whenever a new unknown device is found, where you are asked if you want to connect to it or ignore it for the rest of th [... cut]
closed 2024-03-15 by ImUrX