SlimeVR/SlimeVR-Server #75: Building firmware by server

adigyran, 2021-11-04:
Using [PlatformIO CLI](https://docs.platformio.org/en/latest/core/quickstart.html#process-project) we could build specific firmware by server with just gui interface

Eirenliel, 2021-11-06:
In my idea server uses pre-built firmware for production boards. For DIY it can be useful, yes. It should also read info about currently connected tracker (if it has firmware uploaded) and confure firmware accordingly to its current properties.

TheButlah, 2023-02-12:
@Eirenliel is this a strategy we want? Do we want to add this to v1.0 or should we close it?

Eirenliel, 2023-02-12:
This is not needed in 1.0. As I mentioned before, 1.0 doesn't need firmware update feature since we won't have a new release firmware for a while. Don't need to close it, since this will be needed in the future, but not necesserily built by server.

closed 2024-12-19 by Eirenliel