SlimeVR/SlimeVR-Server #76: Changing board config using server gui

adigyran, 2021-11-04:
Different approach of issue #75, would be having prebuild firmware and  storing defines.h configs in flash with serial commands to change it. But there is a problem with board type, prebuilding for different hardware is not ideal.

Eirenliel, 2021-11-06:
No. Defines are defines, they shouldn't be changed. They *define* the hardware used. Things that should be in the config may be in the config, but defines are hardware-dependant.

closed 2021-11-06 by Eirenliel

deiteris, 2021-11-06:
Except `defines.h` also contains `IMU_ROTATION` and `SECOND_IMU_ROTATION` which have nothing to do with hardware and could be stored either on server or MCU side.

The whole idea is to greatly simplify DIY approach. We don't necessarily need to keep all pinout in flash. We can automate building of `IMU x BOARD` firmwares that people could choose in the server, upload like in https://github.com/SlimeVR/SlimeVR-Server/issues/75 and be on their way without messing around with the code.

ButterscotchVanilla, 2021-11-07:
I'd propose a separate application if you want something to build the firmware, building the firmware would be a niche function to have in the server and a lot of extra code to maintain.

TheButlah, 2021-11-14:
why not just have the rotation be a variable instead of a constant? It can be DEFINEd to a default value, and the server can dynamically modify it via either post-processing on the quats, or by sending a command with the target rotation to the boards and store that in filesystem