SlimeVR/SlimeVR-Server #101: Feature Request: Controller based body proportion setup

fiveseven808, 2022-01-05:
The process of setting up body profiles seems rather tedious and laborious. Can we take a page out of other software such as the those AR piano tutorial things for the quest that locate objects in the real world using the controllers? 

What I'm talking about is, instead of trying to figure out where the trackers are via a lengthy dance, or manual clicking around on the body proportion thing, I would like to use the pointiest end of the controller to define where each tracker is in space on your body. 

Workflow would be like this:
- Define slimeVR trackers per body part/side
- Body proportion setup would start
- Server would ask you to touch the tip (i.e. bottom of quest controller) to the outward facing normal vector of the slimevr tracker, then push a button on the controller in hand. (ex. Touch bottom of quest controller to top of Right Ankle Slime and pull a trigger) 
- Thigh/Ankle/feet trackers can be paired such that the opposing side get automatically assigned symmetrically possibly?
- So long as your feet position doesn't change during this type of calibration, it should work well? 

I know nothing about openvr or what it would take to accomplish this, but if possible, could be super beneficial when sharing slimeVR with other people/guests

Eirenliel, 2022-01-06:
This is something we thought about a few months ago. Problem is, we don't need to know positions of the trackers, we need to know positions of the peoples bones. And it's very dependant on the user's body. It would be very hard to tell user consistently where they should point, minor uncertainties would affect precision, and users can't see their bodies in vr. AutoBone gives the best result so far specifically because it doesn't care about any reference a user can see, it deduces real bones that the software cares about.

I think it's better to work on a good guide to autobone. Doing a few guided movements is faster and easier than locating your body parts in vr, imo, and it would give more precision.

ButterscotchVanilla, 2022-01-07:
> I think it's better to work on a good guide to autobone. Doing a few guided movements is faster and easier than locating your body parts in vr, imo, and it would give more precision.

I agree with this and although AutoBone is quite rough right now, I'm constantly improving it. In the future it'll get much easier to do and more accurate, keep in mind AutoBone should really only need to be run _once_, so however lengthy it may be, it won't be much of a burden.

fiveseven808, 2022-01-07:
> I agree with this and although AutoBone is quite rough right now, I'm constantly improving it. In the future it'll get much easier to do and more accurate, keep in mind AutoBone should really only [...]

For me, VR is a social experience. With that comes people that I fit (vive) trackers to their body when they come over and I want to show them FBT. 

If I had a group of people over, I'd have to use AutoBone for each person which would be very inconvenient and not really practical. If I want to sell people on the technology, it would be good to streamline the sales pitch by minimizing the setup/calibration time. 

If the main concern is figuring out bone length vs tracker position, why not use the controller to locate body points? I.e. from your side (i.e. click on ankles, knees, hip, shoulder, ear) measure distance from headset centerline to controller distance to determine how wide a person's body is. With good instruction/images/animations, I'd imagine it should be no less accurate than what autobone can accomplish. 

If you can use quest controllers to accurately find and position piano keys, I'd imagine it be good enough for what slimevr needs to know.

ButterscotchVanilla, 2022-01-07:
While that is true and I imagine this could be a viable option in the future, I don't think AutoBone should really need much more than a couple seconds of recording to really figure stuff out assuming it will eventually become stable enough. I think this is something other people have looked into a bit and could spend time developing, we'll just have to see with that I guess.

Basically I think this is a good idea, but I'd rather focus on essential functionality for now than extra features.

ButterscotchVanilla, 2022-01-07:
Oh I also forgot to add this on, unfortunately the human skeleton is quite complex and the simplified model used in SlimeVR doesn't perfectly represent that, the advantage that an algorithmic approach can give is that it will take into account those inaccuracies and make adjustments to ensure the best quality using some sort of metric. True measurements are great, but each has its tradeoffs.

closed 2025-03-04 by Eirenliel