SlimeVR/SlimeVR-Server #108: add another protocol USB com port

Carlos1Eduardo1Colpo, 2022-01-16:
instead of connecting via wi-fi, you can use any Arduino with an IMU

ButterscotchVanilla, 2022-01-16:
Would this properly support enough trackers?

Carlos1Eduardo1Colpo, 2022-01-16:
I think so and there are also USB hubs

Eirenliel, 2022-01-17:
This might be added when USB dongle support is implemented

Carlos1Eduardo1Colpo, 2022-01-17:
I think the operating system takes care of these things, just select the input with and it should work, it would be like changing ip just select the correct device

Eirenliel, 2022-01-18:
That's not what I meant. When we implement dongle support we will add serial protocol in firmware and software, so it could be used for this later too.

Kamilake, 2022-02-21:
Obviously, other users can have a separate USB bridge, such as FT232 .
However, since it cannot be added to the support list forever, I want to have the user manually select the tracker in the following ways.

[image]

How about this? Is it OK? That only appears when the tracker is not found and usually works the same as before.

Eirenliel, 2022-02-21:
There will be a serial protocol to distinguish trackers from non-trackers, see #112 so it won't depend on hardware at all

unlogisch04, 2022-08-08:
We should add a filter for "USB Serial Port (COM[0-9]*)" FT232 chip set shows up as this.

ImUrX, 2024-02-01:
We implemented USB-HID support now in #913

closed 2024-02-01 by ImUrX