SlimeVR/SlimeVR-Server #109: Protocol overhaul

Eirenliel, 2022-01-18:
re: discussion in development discord

Plan to overhaul UDP and IPC protocols, maybe using what we learned in implementing new Driver-Server protocol to help build new system that is both backwards and forwards compatible. Plan is to use flatbuffers for ease of development in any language and backwards compatibility.

A few notes:
- Either change port for the new protocol and keep an old one alive for a while
- Or make firmware update system as part of the server to help users migrate to the new protocol
- Not forget owoTrack compatibility (maybe keep old protocol alive specifically for owo)
- Should not overcomplicate build process, users should be able to easily motify and build firmware in platformio without any extra steps or installing extra build tools
- New protocol should be documented well, but also code should be self-explanatory

Eirenliel, 2022-01-18:
Don't work on this yet, please, we will get to it after current refactoring - it won't affect protocol much anyway.

Eirenliel, 2022-01-18:
Alternative to using different port is to use different hello message and add two parsing paths.

TheButlah, 2022-01-18:
Useful: https://google.github.io/flatbuffers/flatbuffers_benchmarks.html

TheButlah, 2022-05-24:
I'm assuming this tracks the firmware protocol rather than the application protocol?

closed 2025-03-04 by Eirenliel