SlimeVR/SlimeVR-Server #112: Improve USB and Serial interface

Eirenliel, 2022-01-26:
Right now all interactions with Serial interface of trackers is done when user presses WiFi button. This should be changed, a new background worker should monitor serial devices and wait for trackers to connect. When it finds new connected tracker, it should start talking to it. A few features that should be implemented:

* Figure out if the connedted device is indeed a SlimeVR tracker
* Make a pop-up that a tracker is connected with suggestion to user to configure it
* If user set auto-configuration, send wifi credentials to it automatically and make a pop-up
* Figure out WHICH tracker is it, and show it as connected via USB in the tracker list (a nice-to-have)

adigyran, 2022-01-26:
oh, i actually want to implement that

unlogisch04, 2022-04-04:
How should we identify the tracker? 
The serial port itself does not have any standard to identify a device. We don't want to mess with existing serial ports. 

I would recommend the following procedure:
1. Open a Configuration mode.
2. Get all serial ports
3. Monitor for new Serial Ports
4. Tell the user to connect the trackers
5. Handshake,...

The following might to be considered:
1. D1 Mini and Most ESP Dev boards use RTS and DTR to Reset or Flash. On some boards they are high on other low. So for DIY the should be able to be changed.

Eirenliel, 2022-04-04:
there is a serial command `GET INFO` that tracker should reply with info about itself

unlogisch04, 2022-04-04:
Well what i wanted to say:
"I'm against sending any Serial Port some characters, without you knowing what it is for."
It can be a old UPS, some other DIY hardware, ...
If we have a other possibility to identify it, like USB native it would be great.

Eirenliel, 2022-04-06:
Then we can wait a few seconds until periodical status message.

unlogisch04, 2022-04-13:
Just for clarification, are your ok with this approach?
I would recommend the following procedure:
  1. Open a Configuration mode.
  2. Get all serial ports
  3. Tell the user to connect the trackers
  4. Monitor for new USB Serial Ports
  5. Open the new found USB Serial Port
  6. Wait for a defined String OR send a String (like `GET INFO\n`)

Eirenliel, 2022-04-13:
I think that new serial devices should always be monitored and automatically connected. If user is not on the Add new tracker screen, they should get a pop up of sorts.

unlogisch04, 2022-04-15:
How often will you connect your tracker to the USB Port to change the configuration?
Once its configured to your network, you will change everything else over the Setting in the Server.
The only reason to connect it to a USB is Charging, WiFi Change, Firmware Update (DIY, not Production)

From looking at the USB Device i have no possibility to know what it is. (i see only a USB to Serial port, with some name)
- If i open it as a serial port, i might reboot a diy 3d printer, or something else.
- If the port is open no other application can access it.

I think the risk of not happy user is too big against the benefit.

Eirenliel, 2022-04-15:
That's fair. Let's make it work only on the Add tracker page.

Kamilake, 2022-08-09:
> * If i open it as a serial port, i might reboot a diy 3d printer, or something else.
> * If the port is open no other application can access it.

That's why I hate cura. That hogs all the com ports and doesn't let go.
My opinion: Exclusive first -> Make sure that is tracker -> Release the other friends

ButterscotchV, 2022-12-16:
> * Figure out if the connedted device is indeed a SlimeVR tracker
> * Make a pop-up that a tracker is connected with suggestion to user to configure it

These two points have been resolved by #312.

TheButlah, 2023-02-12:
@Eirenliel I believe these are all done. Can you confirm, and mark as closed if they are complete

closed 2023-02-12 by TheButlah