SlimeVR/SlimeVR-Server #115: Set device as disconnected in SteamVR when the checkbox is unset

Eirenliel, 2022-02-03:
There is a way to tell SteamVR through driver that the device is disconnected. Idk if it will actually work in games, but we still should use it and notify driver and SteamVR about it when user, for example, unchecks Knees in steamvr config in server.

Louka3000, 2022-06-29:
It *should* work in games. On top of that, we also should translate the tracker to (0, 0, 0). That way we can be sure that VRChat will completely ignore those trackers (it ignores trackers at the origin with IK 2.0).

Louka3000, 2022-07-04:
Update: just marking it as disconnected appears to have the same behaviour as completely turning off a Vive tracker, so messing with position on top is completely useless as this is enough for games (like VRChat) to completely ignore the tracker once marked as disconnected.

closed 2022-07-04 by Eirenliel