SlimeVR/SlimeVR-Server #129: Feature request: Verbose IMU errors

Spazzwanjunk, 2022-02-18:
So im seeing like 50 ppl a day in diy and tech support with "my trackers arent connecting to the server", but it turns out they are connecting but there is an issue with their defines or imu wiring, etc. It would be nice if its possible to have a bit more information.
Suggestion is instead of just not showing up in the server window, it shows up when it can connect but with something like **IMU connection error**. This will help narrow down when it is a problem with the tracker connecting or a problem with just the IMU for easier troubleshooting.

unlogisch04, 2022-02-19:
More Verbose Debug in SlimeVR because of IMU Errors: 
Problem 1: Lot of DIY PPL miss the Tracker in the Server Window. They think the Tracker Wifi has problem and look on the wrong spot.
Possible Solution: Lets still add the Tracker in the Window but show it with either a Error like IMU not Found or IMU Error

Problem 2: Lot of DIY IMU parts come with some different chipset. 6050A but orderd a 6050.
Possible Solution: 
1. Better error description. Device at ID does not match the configured Device in define.h
2. Why not use the IMU-WHOAmiVerifier directly in the FW at boot up to find all possible matches? I agree that this is probably not the best for Production Slimes, so i would suggest to hide this feature behind a debug flag, that normally on in the repo, or active if you use the configuration from the website.
But from this one is also the question. How to get the errors to the Screen. Should we add a server console to the fw, so he can send text data like on serial to the Server? Maybe different Port? (maybe not that easy because we would have to buffer the output from the setup in the firmware, till wifi is available, or wait with the full setup till it is, or a serial console is connected)

Eirenliel, 2022-02-21:
I wanted to add WhoAmI in the firmware, but @Levi--G said it'd be too big of a a code footprint. I still think we should do it, even if without all strings,

closed 2022-02-21 by Eirenliel