SlimeVR/SlimeVR-Server #148: Configuration wizard & Onboarding process

Eirenliel, 2022-04-04:
**Interactive guide through SlimeVR set up and configuration.**

* Wizard automatically opens when user starts SlimeVR for the first time
* Can be launched from the main program screen with a button click (skips Welcome step)
* When multiple languages are supported, each screen of the wizard should have a language selector on the side, for example, top right corner
* If closed during first time set up, should highlight the wizard button. Button should always be highlighted if set up wasn't done at least once (all trackers connected and configured)
* While the wizard is working the reset command from a controller (via feeder app or similar commands from the bridge) may have different actions instead of reset, depending on opened screen: mounting adjustment or starting/stopping Autobone. This will be highlighted in these steps.
* Few of these steps reuse the windows from the main program. This task also doubles as a task for these screens.
* If the user goes back a step or opens the wizard on already configured trackers, all settings should be saved to the best ability, e.g. trackers with assigned roles should keep their assignment, etc.
* If some function is not implemented yet, it still should be in the GUI but marked as unavailable or work-in-progress.

## Steps
Each step can be its own screen or a few steps can be combined into one screen. User can abort configuration during any step. If they open it again, it starts on step 2. The user can always go back a step.

This part tries to describe design details as little as possible, only functional parts, but some advices can be given as an example.

### 1. Welcome
Greets the user if they start SlimeVR for the first time. Encourages them to use the wizard to set up SlimeVR. Empathizes language selector if it exists. 

### 2. Connect all trackers
This step asks user to turn on and connect all the trackers. It needs to show all currently connected trackers, empathize their count (i.e. 5 trackers + 3 extensions should be clearly understood by the user, so they know that they have all trackers connected).

This step should have Wi-Fi config area where user can put wifi credentials. Tracker connected via USB should be automatically configured with this Wi-Fi credentials (#112), connected to Wi-Fi and the server. Any errors that happen during this process should be displayed to the user with an advice on how to fix them, for example, Wi-Fi credentials wrong, unable to find server, busy COM port, etc. This requires implementation of USB serial interface rework issue - #112.

Wi-Fi config area should have a button "Automatically connect other trackers" that will enable Wi-Fi provisioning. See the Wi-Fi provisioning ticket. It tells trackers to find and configure each other's wifi settings while the wizard is open. https://github.com/SlimeVR/SlimeVR-Tracker-ESP/issues/71

Explanation on how to provision trackers via phone should also be included. https://github.com/SlimeVR/SlimeVR-Tracker-ESP/pull/59

The user continues when they think all trackers are connected.

* This step can be re-used as the Add tracker screen. It opens when user clicks "Add tracker" button (should be on the trackers list or on the side panel).

### 3.  Tracker roles assignment
There user is asked to assign trackers to their roles, and put them on.

I see it as a list of trackers on one side, and outline of the body on the other.

A few places on the body are highlighted with names depending on the user's tracker configuration (i.e. for 5 trackers only knees, ankles and chest). The system must to a best guess of user's setup based on the tracker count, but user should be able to change the layout or assign trackers arbitrary. Important parts too keep in mind here are elbow trackers and feet trackers, because they can be swapped - maybe a checkbox "I have elbow trackers" would be useful.

Unassigned trackers in the list are highlighted. When the user clicks one a tracker, they can select on the body where the tracker goes. Or they can click on a drop-down on the side to assign tracker to another place by hand. Assigned trackers display their body part name. Places on the body with the tracker should be visible like occupied. Maybe by moving the tracker to this place, maybe by recoloring it.

During the selecting, tracker's LED should light up for identification (new feature: https://github.com/SlimeVR/SlimeVR-Tracker-ESP/issues/146). User should be informed about it. Trackers should show movement when user picks them up, meaning of this should be explained to the user.

After all trackers assigned, the user can continue. If another tracker added during this process, they should appear in the list and work as normal.

* This step may be reused as a Body view of the trackers in the main window. If "Add tracker" button will be part of this view, here it should return user to the previous step.

### 4. Mounting calibration
Should guide the user through mounting alignment. Show poses which the user should take. See HaritoraX set up process as an example. https://github.com/SlimeVR/SlimeVR-Server/pull/145

It should also tell user that they can invoke alignment from VR, and tell them to launch VR. If the user uses controller to start alignment process it should be two steps, i.e. pressing triggers a few times in one pose, then doing the same in the other. If it's started via button, second pose should be recorded after a timer.

Maybe show two poses and an arrow between them?

* This step can be re-used as Mounting calibration screen.

### 5. Reset
This step should tell user about reset and how to invoke it during the play session. This should explain controller shortcut for reset.

### 6. Body proportions
Can and should be re-used as Body proportions screen.

The differences from the main window may be like:
* The wizard buttons like "Next", "Skip", "Back" etc.
* Some explanation on what's going on
* Always hiding advanced settings like skeleton offsets
* Always hiding irrelevant settings like arms length if there are no elbow trackers
* Can hide body profile selection, but not necessary

**Autobone should be empathized**. Autobone should not be able to start if the user is not in VR. Autobone screen needs improvement:
* Add explanations on what to do
* After data gathering is finished, it should process automatically and display calculated data and an Apply / Discard buttons
	* Calculated data should be sanity checked for ranges and possible wrong values should be highlighted and a warned should be displayed
* Autobone should be possible to start and stop with the controller shortcut for reset. Autobone started from the controller should last until the user uses controller shortcut again.

### 7. Fin!
You're amazing, now go have fun. You can always start the wizard from the main window if you need.

## Other Features

Lsit of features that required for all this to work:

* [ ] #112 
* [ ] https://github.com/SlimeVR/SlimeVR-Tracker-ESP/issues/71
* [ ] https://github.com/SlimeVR/SlimeVR-Tracker-ESP/pull/59
* [ ] https://github.com/SlimeVR/SlimeVR-Tracker-ESP/issues/146
* [ ] https://github.com/SlimeVR/SlimeVR-Server/pull/145

Eirenliel, 2022-04-04:
As mentioned in Discord, Step 2 should have a few clarifications for non-official trackers, like "If you have other types of trackers, connect them now according to their instructions" and maybe links to docs on how to connect owoTrack and JoyCons.

Eirenliel, 2022-04-11:
[image] [Configuration wizard & Onboarding](https://trello.com/c/DzDxUSQr/10-configuration-wizard-onboarding)

ButterscotchV, 2022-12-16:
Mounting calibration is now implemented (though needs some refinement) and a large portion of this has been implemented in the GUI. This issue could use a bit of updating to reflect the current state of things.

ImUrX, 2024-02-12:
This can be kinda closed? the ony thing we never achieved is the fancy wifi provisioning sadly, but it could be another issue!

ImUrX, 2024-07-09:
Superseded by #1102

closed 2024-07-09 by ImUrX