SlimeVR/SlimeVR-Server #177: Feature request: Linux support

yaomtc, 2022-05-21:
Currently SteamVR for Linux is still in "developer release" status, since 2017. "Valve Time" doesn't seem to apply since they haven't announced a date or window to expect a full release. Currently the Store page for SteamVR only shows a Windows icon, and not a SteamOS icon since SteamOS 2.0 did not support VR (not out of the box anyway), SteamOS 3.0 is only intended for the Steam Deck currently, and because of the aforementioned "developer release" status.

https://github.com/ValveSoftware/SteamVR-for-Linux/

I have been using it for a while now, and while it has its issues (see above link), once I'm in-game it does work pretty well.

The Vive Trackers do work on Linux, and since finding this out [I've added them to my list](https://gitlab.com/vr-on-linux/VR-on-Linux). However due to budget I haven't yet picked up a pair for myself.

Anyway, feel free to let me know if you ever decide to try to build on Linux and need a tester.

Eirenliel, 2022-05-22:
This is planned in https://github.com/SlimeVR/SlimeVR-OpenVR-Driver/issues/2 . Server can run on Linux just fine, but there is no driver. I'm closing this issue, since it's a driver problem, not server problem, and is already planned to be fixed eventually.

closed 2022-05-22 by Eirenliel

yaomtc, 2022-05-22:
Thanks, didn't think to check that repo. Would be nice if Github let you search issues across an org's repos. I will keep an eye on that.

ImUrX, 2022-09-21:
@Eirenliel is anything needed on this side? I can PR it, seeing SlimeVR/SlimeVR-OpenVR-Driver#27

kitlith, 2022-09-21:
@ImUrX Yes, that PR needs a corresponding server bridge implementation since the server doesn't support unix domain sockets currently.