SlimeVR/SlimeVR-Server #190: No Blinking of the Tracker if WiFi Window is open / WiFi Window does not show anything if i set the WiFi / Amazon Trackers setting the WiFi

unlogisch04, 2022-06-04:
Problems:
- Some user report that they can not use the WiFi window to set the WiFi credentials. (Hard coding works)
- Some user report that the Tracker stops working if plugged in and they open the WiFi window on the Server

Affected hardware (to my knowledge):
- some Amazon Trackers
- some D1 Mini with CH340c or g USB to Serial chip-sets. (That what i remember from supports)

Workaround:
- Setting up the full Development, And use the workaround:
https://discord.com/channels/817184208525983775/878727840118505533/977740333955821599

Root cause:
- RTS and DTR are used on development boards for Flash Mode and for Reset. But that seems not always to be the same.
- Sometimes they change the state when open the serial connection.

Solution:
- to be defined. 

Dangers:
- Resting the wrong device (3d Printer,... ) if used on the wrong device. 

Open Pull request to allow the user to use change the RTS and DTR to what they like, so they do not need to install everything:
   https://github.com/SlimeVR/SolarXR-Protocol/pull/21
   #186 
 
 Missing:
 - Good description, Advanced Setting
 - New UI integration
 - Help Content
 - Testing on different devices

Eirenliel, 2022-06-04:
After talking in Discord, we found that maybe setting serial control flow to FLOW_CONTROL_DTR_ENABLED may fix the issue, as it has to be enabled for esp to boot into firmware

unlogisch04, 2022-06-04:
After some more writing, 
On the most boards it seems to work when RTS and DTR are set to 0.
Windows seems to default them to 1. it seems some boards do not work that way.

Setting them to 0 by default on our application could help. But the samples i have are too small to be sure that it works in every case.
So i would love to have the option for the user to change this lines. It also gives the possibility to reset the d1 mini from the software to see the full log.
Maybe a survey after the release or a request for feedback would be great to get samples?

Eirenliel, 2022-06-04:
Can we instead make a "RESET" button that does board reset? That would be less confusing, than RTS/DTR checkboxes

unlogisch04, 2022-06-04:
Hm for DIY slime it would work. But for the current review set there seems no HW option for that.
Would that not make it more confusing?

I agree that a "RESTART" button would make more sense. (RESET gets used for Pose)
But somehow i would hope we can hide this RTS/DTR stuff behind advance in the WiFi window. Hidden checkbox that only show when a button Advanced setting is pressed. Even if we have the RESTART button.
Then the user does not need to install all the stuff for only be able to set the WiFi credentials.

TheDevMinerTV, 2022-06-05:
IMO we shouldn't add more features to the old UI as we will get rid of it when we migrate to the new one anyway.
If you want to implement GUI stuff, implement them in [the protocol](https://github.com/SlimeVR/SolarXR-Protocol), [the server](https://github.com/SlimeVR/SlimeVR-Server) and [the new GUI](https://github.com/SlimeVR/SlimeVR-GUI).

unlogisch04, 2022-06-05:
Ok, still missing the solution part. Love to have it in both,
Following changes i would now suggest:
1. Pull RTS and DTR down (set low on code) before opening the Serial Port. (Most driver set them to High at opening)
2. Add options to Set and Reset RTS/DTR over check boxes in a Advance Tab in the "WiFi Window"
3. Reset Button ? how to behave if RTS is set? Toggle it 2 times? toggle both signals 2 times?

How should this switches behave on a new connection? will they save this state and then when you open the port again use the stat before and not 0/0 ?

Eirenliel, 2022-06-05:
> Pull RTS and DTR down (set low on code) before opening the Serial Port. (Most driver set them to High at opening)

How do we do this, down is the default in the lib already

unlogisch04, 2022-06-05:
https://github.com/unlogisch04/SlimeVR-Server/tree/serial_rts_dtr 
When i run the code. The boxes for RTS and DTR come checked. And i just ask the state of RTS and DTR and show it, when after i did open the com port.

unlogisch04, 2022-06-07:
updated :
- https://github.com/SlimeVR/SolarXR-Protocol/pull/21
- https://github.com/SlimeVR/SlimeVR-Server/pull/186

Eirenliel, 2022-06-15:
Agree with Bulah on this: https://github.com/SlimeVR/SolarXR-Protocol/pull/21#pullrequestreview-1008086805

TheButlah, 2022-06-15:
To clarify my position: We should not expose the functionality in the GUI or the protocol, but we should try to automatically solve the DTS/RTS problem for amazon trackers by attempting to retry the serial connection if the initial one didn't work. 

If that proves to be impossible, then yeah we should just not support the misconfigured DTS/RTS

loucass003, 2025-09-24:
@unlogisch04 cleaning up old issues. Is this still a thing? do we still need it?

unlogisch04, 2025-09-24:
I think we can close it. UI was rewriten now years ago.