SlimeVR/SlimeVR-Server #249: Use with other trackers

BeigeAnimal, 2022-09-28:
Is it possible for you to add support with using other SteamVR trackers (Vive trackers, Thundra, etc) as a position reference for slimeVR?
An example is I'm reusing my old Vive wands as flashed trackers, but I want to use SlimeVR for feet, would it be possible for you to add support for SlimeVR to use my trackers as a positional reference for the feet trackers. I'm using the vive wands as knee trackers for context.

Louka3000, 2022-09-28:
duplicate of https://github.com/SlimeVR/SlimeVR-Server/issues/248

closed 2022-09-28 by Louka3000

BeigeAnimal, 2022-09-28:
Oh sorry there was an issue with github when I submited this

Louka3000, 2022-09-28:
> Oh sorry there was an issue with github when I submited this

dw :p
*github is also acting up for me ahhhh*