SlimeVR/SlimeVR-Server #250: Server App not support CH343

SSV-Normandy, 2022-09-30:
I'm building a custom tracker board using CH343 to replace CH340 for smaller size, but the server app can't detect COM with CH343 properly. The COM port is good when download firmware through it.
[image]
[image]

unlogisch04, 2022-10-01:
Should be fixed with
https://github.com/SlimeVR/SlimeVR-Server/pull/245
It will filter for "usb" and "seri"

Louka3000, 2022-10-21:
@SSV-Normandy Can you confirm is this has been fixed (and close the issue if it is)?

Link to a build from latest commit (f3d32da4cd0b5221f0fb25176bf7fe10029faec4): https://github.com/SlimeVR/SlimeVR-Server/suites/8877865892/artifacts/405644841

ButterscotchV, 2022-12-16:
Going to close this issue due to not getting a response, please re-open if #245 did not fix this and it continues to be an issue.

closed 2022-12-16 by ButterscotchV