SlimeVR/SlimeVR-Server #261: Exception in thread "VRServer" java.lang.NullPointerException: Cannot invoke "java.net.InetAddress.getAddress()" because the return value of "dev.slimevr.vr.Device.getIpAddress()" is null

Kamilake, 2022-10-29:
I couldn't find the exact title, so I'll keep the log.
I just ran git pull and got the following exception:
[image]
[image]
```log
22:52:28 [정보] [TrackerServer] Sensor 0 for udp://C8:C9:A3:8C:94:73 status: 1
22:52:28 [정보] [TrackerServer] Added sensor 0 for udp://C8:C9:A3:8C:94:73, type 4
22:52:28 [정보] [TrackerServer] Sensor info for udp://192.168.1.231/0: 1
22:52:28 [정보] [WebSocket] Web Socket VR Bridge started on port 21110
22:52:28 [정보] [WebSocketAPI] New connection from: 0:0:0:0:0:0:0:1
22:52:28 [정보] [SteamVR Feeder Bridge] Pipe \\.\pipe\SlimeVRInput created
22:52:28 [정보] [SteamVR Driver Bridge] Pipe \\.\pipe\SlimeVRDriver created
22:52:28 [정보] [SteamVR Driver Bridge] Pipes are created
22:52:28 [정보] [SteamVR Feeder Bridge] Pipes are created
22:52:28 [정보] [SteamVR Feeder Bridge] Pipe \\.\pipe\SlimeVRInput is open
22:52:28 [정보] [SteamVR Driver Bridge] Pipe \\.\pipe\SlimeVRDriver is open
22:52:29 [심각] Exception in thread "VRServer" java.lang.NullPointerException: Cannot invoke "java.net.InetAddress.getAddress()" because the return value of "dev.slimevr.vr.Device.getIpAddress()" is null
22:52:29 [심각]         at dev.slimevr.protocol.DataFeedBuilder.createHardwareInfo(DataFeedBuilder.java:52)
22:52:29 [심각]         at dev.slimevr.protocol.DataFeedBuilder.createDeviceData(DataFeedBuilder.java:233)
22:52:29 [심각]         at dev.slimevr.protocol.DataFeedBuilder.createDevicesData(DataFeedBuilder.java:283)
22:52:29 [심각]         at dev.slimevr.protocol.DataFeedHandler.buildDatafeed(DataFeedHandler.java:71)
22:52:29 [심각]         at dev.slimevr.protocol.DataFeedHandler.lambda$5(DataFeedHandler.java:120)
22:52:29 [심각]         at java.base/java.util.stream.ForEachOps$ForEachOp$OfRef.accept(ForEachOps.java:183)
22:52:29 [심각]         at java.base/java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:177)
22:52:29 [심각]         at java.base/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:195)
22:52:29 [심각]         at java.base/java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1625)
22:52:29 [심각]         at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:484)
22:52:29 [심각]         at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:474)
22:52:29 [심각]         at java.base/java.util.stream.ForEachOps$ForEachOp.evaluateSequential(ForEachOps.java:150)
22:52:29 [심각]         at java.base/java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(ForEachOps.java:173)
22:52:29 [심각]         at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
22:52:29 [심각]         at java.base/java.util.stream.ReferencePipeline.forEach(ReferencePipeline.java:497)
22:52:29 [심각]         at dev.slimevr.protocol.DataFeedHandler.lambda$4(DataFeedHandler.java:104)
22:52:29 [심각]         at java.base/java.util.ArrayList.forEach(ArrayList.java:1511)
22:52:29 [심각]         at dev.slimevr.protocol.DataFeedHandler.sendDataFeedUpdate(DataFeedHandler.java:103)
22:52:29 [심각]         at dev.slimevr.VRServer.run(VRServer.java:198)
```
<br>
```log
22:57:52 [정보] [TrackerServer] Added sensor 0 for udp://48:55:19:65:92:9F, type 4
22:57:52 [정보] [TrackerServer] Sensor info for udp://192.168.1.38/0: 1
22:57:53 [정보] [WebSocketAPI] New connection from: 0:0:0:0:0:0:0:1
22:57:53 [심각] Exception in thread "VRServer" java.lang.NullPointerException: Cannot invoke "java.net.InetAddress.getAddress()" because the return value of "dev.slimevr.vr.Device.getIpAddress()" is null
22:57:53 [심각]         at dev.slimevr.protocol.DataFeedBuilder.createHardwareInfo(DataFeedBuilder.java:52)
22:57:53 [심각]         at dev.slimevr.protocol.DataFeedBuilder.createDeviceData(DataFeedBuilder.java:233)
22:57:53 [심각]         at dev.slimevr.protocol.DataFeedBuilder.createDevicesData(DataFeedBuilder.java:283)
22:57:53 [심각]         at dev.slimevr.protocol.DataFeedHandler.buildDatafeed(DataFeedHandler.java:71)
22:57:53 [심각]         at dev.slimevr.protocol.DataFeedHandler.lambda$5(DataFeedHandler.java:120)
22:57:53 [심각]         at java.base/java.util.stream.ForEachOps$ForEachOp$OfRef.accept(ForEachOps.java:183)
22:57:53 [심각]         at java.base/java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:177)
22:57:53 [심각]         at java.base/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:195)
22:57:53 [심각]         at java.base/java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1625)
22:57:53 [심각]         at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:484)
22:57:53 [심각]         at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:474)
22:57:53 [심각]         at java.base/java.util.stream.ForEachOps$ForEachOp.evaluateSequential(ForEachOps.java:150)
22:57:53 [심각]         at java.base/java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(ForEachOps.java:173)
22:57:53 [심각]         at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
22:57:53 [심각]         at java.base/java.util.stream.ReferencePipeline.forEach(ReferencePipeline.java:497)
22:57:53 [심각]         at dev.slimevr.protocol.DataFeedHandler.lambda$4(DataFeedHandler.java:104)
22:57:53 [심각]         at java.base/java.util.ArrayList.forEach(ArrayList.java:1511)
22:57:53 [심각]         at dev.slimevr.protocol.DataFeedHandler.sendDataFeedUpdate(DataFeedHandler.java:103)
22:57:53 [심각]         at dev.slimevr.VRServer.run(VRServer.java:198)
```
It happens unconditionally regardless of the latest/old SlimeVR-ESP firmware.
I'm still working on the situation, so if there are further updates, I'll share them right away.

Kamilake, 2022-10-29:
[image]
Handling NullPointerException doesn't seem to be enough at all.

Kamilake, 2022-10-29:
I found that using a newer server with an old improvement GUI caused the following issues

Are there any means or protocols to match the versions of the GUI and the server?

A quick workaround is to exit the new GUI.

(I am trying to handle this exception..)

loucass003, 2022-11-09:
We have been implementing breaking changes recently on the protocol. Because of how early the development of the gui / protocol is. it is going to happen a few times still. We could make it backward compatible but there is no really point doing it rn until the gui gets into production

closed 2022-11-09 by loucass003

Eirenliel, 2022-11-09:
Agreed. Users should use release versions, and for testing whatever version is matching.