SlimeVR/SlimeVR-Server #266: Config loading errors should not lead to server crashing

Eirenliel, 2022-11-11:
Example:

[img](https://cdn.discordapp.com/attachments/1040685976911089796/1040685977015959675/image.png)

Since the solution for the user is pretty much always to delete the config file, we should do either of the two things:
* Ignore error and load what loads
* Ask user if they want to ignore error and potentially reset part of the config, or close the server and they can try fix the config by hand

Louka3000, 2022-11-15:
note: the exception in the screenshot has been fixed and merged in `main`.

TheDevMinerTV, 2022-11-15:
option 3: if it doesn't load, back up the  config and reset to default settings

Eirenliel, 2022-11-15:
yes, it's also good option

ButterscotchV, 2022-12-16:
Marking this as high priority since it can "mysteriously" completely break the functionality of the server if a mistake is made either in the server or by the end user.

> option 3: if it doesn't load, back up the config and reset to default settings

I think this is a good option, we should at a minimum do this and warn the user.

ButterscotchV, 2023-01-16:
Partially implemented via #476, what's left to be done is GUI functionality, still booting the server even with a config error (to forward it to the gui), and allowing the user to reset the config simply with a dialog choice.

I've discussed this issue on Discord and my conclusions were in this message (though my PR doesn't *exactly* follow this in order to always atomically back up the config on loading failure): https://discord.com/channels/817184208525983775/948579740225261598/1054723637673996350

> Success write case:
> - Write `vrconfig.yml.tmp`
> - Rename `vrconfig.yml.tmp` to `vrconfig.yml` to overwrite
> 
> Failure write case:
> - Write `vrconfig.yml.tmp`
> - Fail (either at write or at renaming)
> - Report to user
> 
> Failure load case:
> - Fail to read `vrconfig.yml`
> - Inform user and give option to reset config
> - If reset:
> - Try to rename `vrconfig.yml` to `vrconfig.yml.bak` and ignore if failed
> - Write `vrconfig.yml.tmp`
> - Rename `vrconfig.yml.tmp` to `vrconfig.yml` (standard write)

marcozzxx810, 2023-02-06:
@ButterscotchV, as per our conservation in discord. The GUI still crashes in its current state and we would like to make the below modification to show the config error:

- [ ] send the config error to GUI
- [ ] show GUI modal or popup to notify the user 

For the popup/ modal, apart from stating there is an error, any action that the user should perform (Eg. a button for reset the config to default, or a button to review the old config)

closed 2025-09-24 by loucass003