SlimeVR/SlimeVR-Server #270: Allow temporary "pausing" of tracker info sent to SteamVR

nullstalgia, 2022-11-14:
Hi! (Especially with the upcoming merge, this feels more relevant here) I would like to request a binding/button/switch on the GUI that will temporarily stop sending tracker updates to SteamVR.

When using SlimeVR and I need to go out of the room or use the washroom, my having my trackers still going and broadcasting my movements isn't always appreciated (especially with the latter example). Or if I'm doing a mounting reset and don't want people to see me do weird leans for a bit if I mess it up.

So having a way to keep stuff like the SlimeVR skeleton working while giving SteamVR static data for a bit would be ideal, I think.

ButterscotchV, 2022-11-20:
Agreed on this, it would be incredibly simple to implement by just having the skeleton able to be disabled. When disabled, it just simply wouldn't run the update loop.

ImUrX, 2023-06-29:
Added by #697

closed 2023-06-29 by ImUrX