SlimeVR/SlimeVR-Server #276: Selectable serial port

ButterscotchV, 2022-11-20:
Currently the serial port just automatically connects to one, we should have some sort of dropdown to select which we want to connect to.

TheButlah, 2022-11-22:
@ButterscotchV is this GUI or server? Please label area

Chemrat, 2022-11-22:
Both GUI and server are missing this option.
On devices that for some reason expose multiple serial ports, this makes impossible to change wifi credentials and do initial tracker setup.
I have managed to set up the wifi manually, as a workaround, by typing in serial console (e.g. PlatformIO serial monitor window) `SET WIFI "ssid" "passwd"`

TheButlah, 2022-11-23:
Why would a device have multiple serial ports? Shouldn't the server manage this in a way entirely hidden from the user?

@unlogisch04 maybe you can offer some insight? Why would we need to select a serial port

unlogisch04, 2022-11-23:
> Why would a device have multiple serial ports? Shouldn't the server manage this in a way entirely hidden from the user?
Well each tracker will not have more than 1 Serial port. But if you plugin more than 1 tracker, you will have more than 1 serial port. Add Wavecom, 3D Printer,... to it. 

So the question is, how to know what is a tracker and what not.
For ESP32-C3 you can somehow get a HW ID. I'm not sure if you do that, that you would also have to write UBS Drivers for it. Or if you still use the normal USB to Serial drivers. 

As Chemrat stats, the server just try to open the first Serial Port. If that fails it just try it again. You never know to what device the server exactly will connect.

I don't have time to programm any of this. 

I have the following question. Where should the Com Port be select able?
The WiFi Window no longer exists) and The Wizard don't seem to be a good option to add a drop-down. 

I did come up with the following idea:

For port selection: (Serial Monitor)
1. If some application(GUI) requests it, Pull all Serial Ports with they name. Send this list to the (GUI)
3. User select a Port: GUI -> Server sends a Request to open that port

For tracker configuration: (credentials)
1. Automatic WiFi assignment (Wizard Mode) GUI requests Wizard Begin Mode
2. Server gets a list of all actual connected Serial Ports - Server sends ready to GUI
4. Server listen to new connected Serial Ports / USB Devices -> If it is a Serial Port, Open it, Listen for Text of SlimeVR (notify GUI)
5. notify GUI  -> found new device and ask if to set WiFi Credentials  GUI -> Server Yes/No (this I'm not sure if needed)
6. Server send the WiFi data to the tracker -> Check if successful -> disconnect  (notify GUI)
7. GUI -> can abort this listening, Server -> aborts if no more connected to GUI or after 5 minutes?

Needed GUI:
- Changes on Serial Monitor
- Changes on Wizard? @loucass003

TheButlah, 2022-11-23:
Cant `esptool` and `espflash` easily detect attached esps? 

I would much rather the server handle specific details like serial ports - the gui just wants to see logs from the firmware. Those could be over serial, over wifi, over JTAG - who knows!

unlogisch04, 2022-11-23:
> Cant `esptool` and `espflash` easily detect attached esps?
https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/basic-options.html
`If no -p option or ESPTOOL_PORT value is specified, esptool.py will enumerate all connected serial ports and try each one until it finds an Espressif device connected.`

TheButlah, 2022-11-23:
One thing that needs to be decided - we *could* just have the user type into a textbox the serial port, but its often nicer to have a list of options for them to pick by enumerating the serial ports.

But this is hard because then basically the server would need to tell the GUI about the ports available as options.

Perhaps we could reconcile this by having the server tell the gui the list of *possible* firmware log sourcces? These could be serial ports, networked logs, etc. So by default we would report all serial ports, and then maybe even names for each "network log" source (these network logs don't exist yet, but they might in the future).

That lets the gui populate a dropdown with these options, and when they click on one, the gui sends it to the server.

Eirenliel, 2022-11-23:
Server should send GUI the list of serials it found, and currently active one (if any), and user should be able to select which to use.

> But this is hard because then basically the server would need to tell the GUI about the ports available as options.

Why is this *hard*, it's basic function of things, give a list of options...

TheButlah, 2022-11-23:
Its not hard its just not as easy as "gui sends a string to server". Now we need server to send events to gui as part of the process too, there must be a task that continuously runs in the server to constantly enumerate ports, must publish that info, etc.

TheDevMinerTV, 2022-11-23:
> Cant `esptool` and `espflash` easily detect attached esps?
> https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/basic-options.html
> `If no -p option or ESPTOOL_PORT value is specified, [...]

sadly this isn't that easy, it still finds my Rift S as a device.

Eirenliel, 2022-11-23:
> Its not hard its just not as easy as "gui sends a user submitted string to server". Now we need server to send events to gui too, there must be a task that continuously runs in the server to constantly enumerate ports, must publish that info, etc. And then we still need to implement the RPC for the gui sending the user-chosen string to the server.

so happy i'm still useful in this project, my role is to keep repeating "this was planned for a long time!" stuff like auto-detecting connected slimes to usb was alwasy planned, even automatical wifi provisioning to them.

this is really trivial! i understand the desire to limit everything to just your scope, but server and GUI are the same thing, especially right now. features should be done the best way, not the smallest code impact way.

unlogisch04, 2022-11-23:
Before i forget again  `Improve USB and Serial interface` #112

ButterscotchV, 2022-12-16:
Seems to have been resolved by #312, please re-open otherwise.

closed 2022-12-16 by ButterscotchV