SlimeVR/SlimeVR-Server #277: Beta distribution

ButterscotchV, 2022-11-21:
I would love to be able to use GitHub Actions for sharing beta builds since it's a trustworthy source and not just some random jar off my computer, BUT it's impossible for people without GitHub accounts to download it, I've looked around and found this hacky workaround service for getting direct links, *but* it's adding another service we need to trust to it...
https://nightly.link/

As an alternative, I think it's *viable* to keep sending files directly in the Discord to be publicly available, but I think we should modify the CI to include the hash in the zip name (or somewhere else easily accessible) in order to verify downloaded files either from the service mentioned above, most people won't make use of this, but I think it'd be worthwhile to ensure nobody is sneaking stuff in

This could be relevant to the auto-updater in the future for downloading beta/nightly versions and validating the integrity of the builds

ButterscotchV, 2022-12-16:
As an update to this, we now have a [\#beta-testing](https://discord.com/channels/817184208525983775/1047570753001885748) channel on the Discord to aid in the distribution. This seems to have done pretty well in improving the organization and I think it's a good direction to continue in.

Louka3000, 2023-06-14:
We are now using Github pre-releases for release candidates and Discord forums for all betas.

closed 2025-03-04 by Eirenliel