SlimeVR/SlimeVR-Server #281: Adding a "Help" tab

ButterscotchV, 2022-11-22:
It would be of benefit to have some sort of tab on the GUI for help, be it where to find help (the docs or the Discord) or for a potential feedback survey.

loucass003, 2022-11-27:
We need to really state all the stuff we want on this page. this is still too vague to be sent to mrjvs imo

nullstalgia, 2022-11-27:
It could be a glorified link page, with different buttons linking to articles on the Wiki. Or there could be small tips built into the app (like common fixes, like USB drivers, trying different ports/cables, see if the LED is blinking, etc) that don't require an internet connection.

What sounds best to you?

ButterscotchV, 2022-11-27:
I think it would be good to have just a bunch of links for now, any specific help stuff could be displayed on the GUI wherever it's possible

I think it should be something like:
Docs Common Issues page
SlimeVR Discord
Maybe SlimeVR GitHub?

ImUrX, 2023-07-26:
There is currently a link to docs as a question mark!