SlimeVR/SlimeVR-Server #283: Have pictures for Autobone

Louka3000, 2022-11-22:
- [ ] Figure out which poses are best for autobone (sitting in chair, squatting, bending over, etc…) @ButterscotchV can make that list! :p

- [ ] Draw several pictures of a humanoid doing these poses.

- Alternatively, some kind of animation (gif?) showing each step one after the other could work.

- [ ] This then needs to be integrated into the new GUI in the setup wizard in the autobone section to show the user which moves to do while autobone is running.
- note: we may also want a neutral/idle standing pose to tell the user what to do before autobone starts running.

ButterscotchV, 2022-12-16:
I have been working on this and discussing it on Discord with others, this is the current latest reference for the poses and such, I still want to work on it a little more, so I won't dedicate to these yet, but it's the best I have so far.
[image]

Eirenliel, 2023-02-20:
As we discussed with @loucass003 before, here's a bit of a ToDo to change *current AutoBone version* without big changes.

## Page 2
* [ ] Remove page 2 completely
[image]

## Page 3
Rework page 3.
[image]

* [ ] Remove the warning. Why is it here?
* [ ] Change text to: "We're now going to record some poses and moves. You can see the moves on the right. Recording will start when you press the button. **Be sure to keep your feet at the same place and heels planted while doing the moves!**
* [ ] Add slideshow of the moves that Butter posted above to the right of the page where it's empty right now. The slideshow should start with "Don't move your heels", there should be a slider showing how many slides there are and what slide it's displaying now.
* [ ] Pressing "Start recording" starts recording, ON THE SAME PAGE.
* [ ] After recording is started the button is turned inot "Abort recording"
* [ ] If Abort recording is pressed, no information is sved or processed
* [ ] After recording is started additional text is displayed above the buttons: "Recording in progress, do the moves!"
* [ ] After recording is started, pose sliders are reset to slide 1
* [ ] After recording is finished, move to the next page with proportions

## Page 4
* [ ] Page 4 is removed and integrated with Page 3
* [ ] The button "Redo recording" returns on the previous page, but doesn't start recording right away

ButterscotchV, 2024-07-19:
I think this is basically done now? Can this be closed

ButterscotchV, 2024-09-17:
Going to close this as it should be completed. Please re-open this or make a new issue if there is anything left to do.

closed 2024-09-17 by ButterscotchV