SlimeVR/SlimeVR-Server #293: GUI detection of motion is not sensitive enough

TheButlah, 2022-11-24:
I think that the "motion" outline on the gui should:
1. Be binary, not brighter/darker depending on how much motion
2. Should be a lot more sensitive

I tend to find that its not sensitive enough, I have to *really* move the trackers when I do it.

nullstalgia, 2022-11-24:
I think it only checks for movement on one axis? Since when I was assembling my BNO085 carrier boards, i had to wave them long-wise to get it to light up. That, or that was just the only axis I was able to get it fast enough with. 🤔

Louka3000, 2022-11-24:
Would using the accel work? So people can move their trackers without rotating them and it'd register movement...

nullstalgia, 2022-11-24:
Hmm, that could also work well. As long as you keep the gyro sensing as well since not everyone has accelerometers in their DIYs.

TheButlah, 2022-11-24:
I think both should be used

closed 2023-02-04 by loucass003