SlimeVR/SlimeVR-Server #305: Server should send the HMD transform used at the time of FK solving to the steamvr driver

TheButlah, 2022-11-25:
Discord thread: https://discord.com/channels/817184208525983775/1042662909043544085/1042662909043544085

Right now, there is a noticeable lag that can be observed from slimevr, especially when dancing with a chest tracker turned on.

Slime has some inherent latency, and this latency causes the chest tracker to lag behind the HMD tracker, causing the neck of the avatar to stretch and compress in a really noticeable way, as well as lean.

We should use the relative translation between the head transform sent by the server and the head position in world space at the time of reception to apply a translation to *all* of the tracker nodes, so that they smoothly follow the latest HMD data.

TheButlah, 2022-11-25:
In order to accomplish this, I believe we need to change the protobuf protocol in the openvr bridge to have *all* trackers sent in one packet, rather than one packet per tracker. This will allow us to know which tracker transforms to update when we find this new relative head transform.

maemoKKK, 2023-12-19:
I think we should abandon openvr ASAP.

I don't know if some people with ulterior motives are continuing to recommend OpenVR