SlimeVR/SlimeVR-Server #308: Relicense SlimeVR under Dual MIT + Apache 2.0 license

TheButlah, 2022-11-25:
With #306 and the discussion on discord, it seems like we are approaching a consensus that people would be willing to dual license.

# What is the proposed change?
SlimeVR will be dual licensed under both [LICENSE-MIT](https://github.com/SlimeVR/SlimeVR-Server/blob/e5c05fc8644b6ff9e0a6b3d381d904bdcf971447/LICENSE-MIT) and [LICENSE-APACHE](https://github.com/SlimeVR/SlimeVR-Server/blob/e5c05fc8644b6ff9e0a6b3d381d904bdcf971447/LICENSE-APACHE).

All contributions will be licensed such that we grant users of our software the right to choose to use either license. So if I want to use the MIT license, I can - if I want to use Apache 2.0, I can.

# Why Apache 2.0?
MIT license doesn't protect against patent trolls. Technically SlimeVR could get a patent on fbt and then sue people for patent infringement. We want to give users the confidence that this won't happen.

For this reason, other libraries like the GUI, SlimeVR-Rust, SolarXR-Protocol, SlimeVR wrangler use dual licensing.

# Why do we care enough to adopt this?
To keep the server repo's licensing simple, it is going to be confusing for the GUI to be dual licensed but the server not to be. Lets just unify everything under a dual license.

# How do we do it?

I will follow the [example of other libraries I know](https://github.com/bevyengine/bevy/issues/2373).
I will @ every contributor to SlimeVR to get their agreement to relicense. Assuming everyone (who has done a non-trivial change that can't easily be reverted) agrees, we can officially relicense and unify the codebase under dual MIT/Apache 2 licenses.

# Agreement
Paste the following into a comment in this thread:
> I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked as "Eiren Rain and SlimeVR Contributors".

- [ ] @Eirenliel 
- [ ] @ButterscotchV 
- [ ] @loucass003 
- [ ] @Louka3000 
- [ ] @TheButlah 
- [ ] @Spazzwanjunk 
- [ ] @adigyran 
- [ ] @deiteris 
- [ ] @TheDevMinerTV 
- [ ] @carl-anders 
- [ ] @ColdIce1605 
- [ ] @Kamilake 
- [ ] @Stermere 
- [ ] @JimWails 
- [ ] @williammb 
- [ ] @ImUrX 
- [ ] @unlogisch04 
- [ ] @Loler920a 
- [ ] @Blixtdraken 
- [ ] @MightyGood
- [ ] @LETS-BEE
- [ ] @Boronare
- [ ] @Levi--G
- [ ] @masterex1000
- [ ] @Vyolex
- [ ] @JINODK
- [ ] @Timocop
- [ ] @doormatt-dev
- [ ] @ThreadOfFate
- [ ] @sctanf
- [ ] @0x8080
- [ ] @ThePheonixGuy
- [ ] @nullstalgia
- [ ] @bluekani
- [ ] @inkuxuan
- [ ] @DemorianJH
- [ ] @gbpii
- [ ] @Charlito33
- [ ] @ChlorideCull
- [ ] @nEuDy
- [ ] @Torikari
- [ ]

TheButlah, 2022-11-25:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked as "Eiren Rain and SlimeVR Contributors".

ImUrX, 2022-11-25:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked as "Eiren Rain and SlimeVR Contributors".

carl-anders, 2022-11-25:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked as "Eiren Rain and SlimeVR Contributors".

TheDevMinerTV, 2022-11-25:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked as "Eiren Rain and SlimeVR Contributors".

sctanf, 2022-11-25:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked as "Eiren Rain and SlimeVR Contributors".

Stermere, 2022-11-25:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked as "Eiren Rain and SlimeVR Contributors".

ThePheonixGuy, 2022-11-25:
I agree that any contribution intentionally submitted for inclusion in the
SlimeVR github organization or its repositories, shall be dual licensed
under MIT License and Apache 2.0 license, as described above, without any
additional terms or conditions. I agree that the copyright owner shall be
marked as "Eiren Rain and SlimeVR Contributors".

Eirenliel, 2022-11-25:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked as "Eiren Rain and SlimeVR Contributors".

ButterscotchV, 2022-11-25:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked as "Eiren Rain and SlimeVR Contributors".

Louka3000, 2022-11-25:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked as "Eiren Rain and SlimeVR Contributors".

MightyGood, 2022-11-25:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked as "Eiren Rain and SlimeVR Contributors".

kitlith, 2022-11-25:
~~i don't see my name listed but eh~~

I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked as "Eiren Rain and SlimeVR Contributors".

unlogisch04, 2022-11-25:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked as "Eiren Rain and SlimeVR Contributors".

TheButlah, 2022-11-25:
I changed the text to the following:
> I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked either as "Eiren Rain and SlimeVR Contributors" or "SlimeVR Contributors".

This is because Eiren may wish to remove their name.

Vyolex, 2022-11-25:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked either as "Eiren Rain and SlimeVR Contributors" or "SlimeVR Contributors".

Spazzwanjunk, 2022-11-25:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked either as "Eiren Rain and SlimeVR Contributors" or "SlimeVR Contributors".

Blixtdraken, 2022-11-25:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked either as "Eiren Rain and SlimeVR Contributors" or "SlimeVR Contributors".

Charlito33, 2022-11-25:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked either as "Eiren Rain and SlimeVR Contributors" or "SlimeVR Contributors".

JINODK, 2022-11-25:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked either as "Eiren Rain and SlimeVR Contributors" or "SlimeVR Contributors".

ThreadOfFate, 2022-11-25:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked as "Eiren Rain and SlimeVR Contributors".

williammb, 2022-11-25:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked as "Eiren Rain and SlimeVR Contributors".

loucass003, 2022-11-25:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked either as "Eiren Rain and SlimeVR Contributors" or "SlimeVR Contributors".

deiteris, 2022-11-25:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked either as "Eiren Rain and SlimeVR Contributors" or "SlimeVR Contributors".

nullstalgia, 2022-11-25:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked either as "Eiren Rain and SlimeVR Contributors" or "SlimeVR Contributors".

Loler920a, 2022-11-26:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked either as "Eiren Rain and SlimeVR Contributors" or "SlimeVR Contributors".

Kamilake, 2022-11-26:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked either as "Eiren Rain and SlimeVR Contributors" or "SlimeVR Contributors".

doormatt-dev, 2022-11-26:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked as "Eiren Rain and SlimeVR Contributors

inkuxuan, 2022-11-26:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked as "Eiren Rain and SlimeVR Contributors".

bluekani, 2022-11-26:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked either as "Eiren Rain and SlimeVR Contributors" or "SlimeVR Contributors".

masterex1000, 2022-11-26:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked either as "Eiren Rain and SlimeVR Contributors" or "SlimeVR Contributors".

ColdIce1605, 2022-11-26:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked either as "Eiren Rain and SlimeVR Contributors" or "SlimeVR Contributors".

JimWails, 2022-11-26:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked either as "Eiren Rain and SlimeVR Contributors" or "SlimeVR Contributors".

Timocop, 2022-11-26:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked either as "Eiren Rain and SlimeVR Contributors" or "SlimeVR Contributors".

LETS-BEE, 2022-11-26:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked either as "Eiren Rain and SlimeVR Contributors" or "SlimeVR Contributors".

unlogisch04, 2022-11-26:
Sorry for the Ping.
Due to the change i had to remove the checkmkark from some user that already did replay but did not update the comments.
See comment from @TheButlah https://github.com/SlimeVR/SlimeVR-Server/issues/308#issuecomment-1327792019

So following user have to update the comment to agree on the new text. (Post the new text)

@carl-anders
@doormatt-dev
@ImUrX
@inkuxuan
@kitlith
@Louka3000
@Stermere
@ThePheonixGuy
@ThreadOfFate

ImUrX, 2022-11-26:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked either as "Eiren Rain and SlimeVR Contributors" or "SlimeVR Contributors".

Louka3000, 2022-11-26:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked either as "Eiren Rain and SlimeVR Contributors" or "SlimeVR Contributors".

Levi--G, 2022-11-26:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked either as "Eiren Rain and SlimeVR Contributors" or "SlimeVR Contributors".

DemorianJH, 2022-11-26:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked either as "Eiren Rain and SlimeVR Contributors" or "SlimeVR Contributors".

abb128, 2022-11-26:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked either as "Eiren Rain and SlimeVR Contributors" or "SlimeVR Contributors".

Boronare, 2022-11-28:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked either as "Eiren Rain and SlimeVR Contributors" or "SlimeVR Contributors".

adigyran, 2022-11-28:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked either as "Eiren Rain and SlimeVR Contributors" or "SlimeVR Contributors".

TheButlah, 2022-11-28:
@gbpii you need to say "Eiren Rain and SlimeVR Contributors" or "SlimeVR Contributors" you only wrote one

Stermere, 2022-11-28:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked either as "Eiren Rain and SlimeVR Contributors" or "SlimeVR Contributors".

Torikari, 2022-11-30:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked either as "Eiren Rain and SlimeVR Contributors" or "SlimeVR Contributors".

nEuDy, 2022-12-01:
I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as described above, without any additional terms or conditions. I agree that the copyright owner shall be marked either as "Eiren Rain and SlimeVR Contributors" or "SlimeVR Contributors".

TheButlah, 2022-12-02:
@ChlorideCull submitted [this](https://github.com/SlimeVR/SlimeVR-Tracker-ESP/pull/76/files) commit which does would not fall under copyright law as a creative work, so I have marked their name as checkmarked.

TheButlah, 2022-12-02:
Same with @0x8080: https://github.com/SlimeVR/SlimeVR-Tracker-ESP/pull/24/files

Louka3000, 2022-12-05:
@ThePheonixGuy Can you edit your comment or add a new one with the updated text please
> I agree that any contribution intentionally submitted for inclusion in the SlimeVR github organization or its repositories, shall be dual licensed under MIT License and Apache 2.0 license, as [...]

ThePheonixGuy, 2022-12-26:
I agree that any contribution intentionally submitted for inclusion in the
SlimeVR github organization or its repositories, shall be dual licensed
under MIT License and Apache 2.0 license, as described above, without any
additional terms or conditions. I agree that the copyright owner shall be
marked either as "Eiren Rain and SlimeVR Contributors" or "SlimeVR
Contributors".--
James Reynolds

Senior Platform Engineer

***@***.***

TheDevMinerTV, 2022-12-26:
Done! Finally!

closed 2023-01-18 by TheButlah