SlimeVR/SlimeVR-Server #318: OpenXR Support

ButterscotchV, 2022-11-29:
OpenXR support for SlimeVR would be nice for future-proofing and extended compatability.

ImUrX, 2023-03-03:
This is blocked, there is no API from OpenXR 1.0 for adding a device. This is what I found:
> OK, so that sadly isn’t supported yet. It’s on the roadmap and is called the device plugin for now. Defining a plugin interface for 3rd party devices (input, HMDs etc) on various platforms (not just OSes, but also mobile vs. desktop and VR vs AR) is a complex problem if you want to get it right in version 1. So it had to be delayed from OpenXR 1.0.
I can’t give you an ETA for that feature sadly, but delayed does not mean cancelled

ROBYER1, 2023-03-03:
> 

I assume this post was from you? Fingers crossed we get some update, I'll ask around too 
https://community.khronos.org/t/create-openxr-device-driver-custom-runtime/109475