SlimeVR/SlimeVR-Server #325: Have seperate toggles for sending and receiving OSC?

Louka3000, 2022-11-30:
Discussion: should we?
- This would make it so users don’t have to worry about clogging ports.
- Might add some confusion tho.
- Also maybe too late to easily do it because of protocol?

Louka3000, 2022-12-01:
@TheButlah @loucass003

TheButlah, 2022-12-01:
by toggles, do you mean should we have both the steamvr and osc sections?
[image]
[image]

I didn't understand what you were referring to.

Louka3000, 2022-12-01:
No, I making the single OSC "Enable" toggle into 2 "Enable sending" and "Enable receiving" toggles. Nothing to do with SteamVR.

TheButlah, 2022-12-01:
Too many settings bad for users. I'm not worried about hogging another port, or sending a little bit of extraneous data that will never arrive

ButterscotchV, 2022-12-06:
Could be nice if it's hidden behind advanced settings or whatever, keep it simple for the user unless they specify that they want more

TheButlah, 2022-12-07:
I would agree if it actually did anything useful - but I don't think it really does

Louka3000, 2022-12-07:
Users **could** just put a random number in, for example, "Port In" if they don't wanna receive data.
But at the same time, that could potentially break other programs (altho probably extremely unlikely), and it's not a pretty user-friendly way to do it IMO.

Also need to consider when users would not want to receive data ig... Support for stuff like https://github.com/SlimeVR/SlimeVR-Server/issues/323 would just not have any OSCSender (thus no Port Out in GUI)... There might not be any situation where SlimeVR allows to user to receive or send data while the user wouldn't want one of those...

TheButlah, 2022-12-07:
protocol != GUI
GUI can have a value of -1 and do something with the protocol.

Louka3000, 2022-12-07:
Yes but that would require having 2 booleans in the protocol. I'm just saying it can't work with the current protocol and GUI.

Louka3000, 2023-03-08:
Doesn't seem like this is interesting, as clogging up a port probably isn't a concern

closed 2023-03-08 by Louka3000