SlimeVR/SlimeVR-Server #393: Add sentry.io error tracking

ImUrX, 2022-12-15:
I need opinions on others if this is a good idea, this basically sends errors to a server where we can see errors happening to the code, it works really well (it includes build info, backtracing and a lot more we can add ourselves) and it can be self-hosted.
This would require a Privacy Policy, the idea would be to have it disabled by default and add an opt-in option in the GUI somewhere, this would link to the privacy policy so the user can read it.

This would help a lot for finding errors that aren't actually getting reported by users and have a better tracking on the precise errors.
I don't find any cons tbh because it's disabled by default and it will be pretty helpful if it's actually enabled.

TheButlah, 2022-12-15:
Does sentry.io have good java integration? What about rust and typescript? can we use this same service across all parts of the tech stack?

TheDevMinerTV, 2022-12-15:
[image]
compatibility shouldn't be an issue 😅

ImUrX, 2022-12-15:
2. It's open source
3. You can pay [sentry.io](https://sentry.io/pricing/) or self-host it, we can try it before actually paying for it :P
4. Maybe New Relic? But like sentry.io is open-source
5. It's not automatic, it wont log errors that are catched. We need to add everything ourselves. It has a lot of features like breadcumbs that would help us give custom information for context-sensitive errors also

TheDevMinerTV, 2022-12-15:
New Relic is hella expensive

cleafspear, 2024-06-07:
regarding as someone who is a active user with sentry:
sentry has multiple pay tiers. they limit the amount of ingested data normally based on the pay tier you use, but the self-hosted edition does not have any limits on it, and is not cost, at the limitation that you will have community support for issues.
in regards to what data it collects? its entirely based on your error handler. you can have it provide anything you want in the logs, tho by default, it normally contains the following:
os type; hardware stats (cpu,gpu,ram) log level :error,warning,ect.,machine ID (only to match with other errors from the same machine, and a user UUID (only useful to id machines logged into with the exact same user/domain joined machines)
it is very useful to id what is going on and can give a statistic on what is causing the errors.

closed 2025-03-27 by Eirenliel