SlimeVR/SlimeVR-Server #398: Switching the codebase to Kotlin

ButterscotchV, 2022-12-16:
This is a proposal to switch the server codebase to Kotlin! I'd like to guage the general opinion on this and what might be any pros/cons about switching.

I propose for the switch, we can do each part of the server without an active PR or active development first, then slowly as PRs clear up, we can switch those remaining classes to Kotlin without loss of development time or huge merge conflicts. This switch should be fairly easy and for the most part should be fully automated.

Pros:
 - Non-nullable by default. This provides way better protection against null pointer exceptions.
 - Less verbose, more hinting. This reduces the overall size of the code and makes it much easier to look through and review. When there's less to look at, the more likely you are to spot problems.
 - Loads of shortcuts for ease of use. For example, replacing a full line of a null check to just `?.`.

Cons:
 - Would be a fairly long transition from Java to Kotlin for the codebase.
 - Everyone would need to learn how to use Kotlin, though it is fairly similar to Java.
 - Reviews may be a bit slower as people get used to the change.
 - May not work with all the tools that we currently use (ex. Spotless maybe?).

You can read more about Kotlin here: https://kotlinlang.org/docs/home.html

TheButlah, 2022-12-16:
We discussed this in discord for a while. Here is a summary:

So we discussed this in discord more. Here is a summary, which I ran past the main participants of the conversation which were @TheButlah @Louka3000 @ButterscotchV @ImUrX to make sure I didn't miss something.

**Motivation for using Kotlin:**
*  **The primary motivator was the null safety of kotlin.**
* We observed numerous NPE issues in the last month alone that broke the server. Theory is that kotlin would eliminate or reduce these.
* It was also discussed that kotlin was less verbose than java, and had nice operators like ? and did var by default.
* There was some discussion of kotlin having an easier concurrency story than java with the use of coroutines and promises.
* There was less discussion about other reasons for use of kotlin, but it seemed that most people who participated in the conversation were more interested in kotlin than java or at least curious/open to the idea.

**Ability to incrementally port to Kotlin:**
* Kotlin lets you port the code on a per-file basis.
* You get to call java from kotlin and vice versa as a first class citizen. So no FFI boilerplate or anything is needed.
* Both can be compiled into same library, same jar, etc.
* There are tools to autoconvert from java to kotlin. This makes it really easy as most of the work can be automated.
* We decided on a good workflow: Pick a class at high risk of null issues. Autoconvert its file to kotlin. Go through and manually clean it up or make the nullability more strict. Submit a PR.
* **We ended up with a consensus that it is easy to incrementally port to kotlin. No one appeared to disagree with the ease of porting the code incrementally.**

**Rewrite vs Kotlin:**
* Some felt that they preferred a rewrite in another language (go and C# were the primary ones mentioned) over something incremental. Others felt that they preferred incremental porting to kotlin instead of a rewrite. Those who wanted a rewrite felt that kotlin didn't quite solve all the issues they had with java, so they wanted to go even further.
* We observed that slimevr ships soon so the existing code must continue to receive features and bugfixes until a rewrite has reached the same level of maturity. Whereas doing something incremental means we don't need to steal resources from the main code while the rewrite happens.
* We felt we may lose contributors if we switched to golang, although some felt it wasn't hard to use.
* Some felt that c# would be an easy thing for contributors to switch to, but one contributor was pretty against it. Seemed to all agree that C# had more language bloat than java though, but people disagreed on whether that actually mattered.
* **We reached a consensus that we thought that it would be far easier to use kotlin instead of a full rewrite to a completely separate language. No one seemed to disagree with the fact that incremental use of kotlin would be easier than a full rewrite. Those who wanted a full rewrite still preferred incremental Kotlin over keeping it Java-only.**

**Concerns about whether the contributors would be ok with it**
* We all were concerned if others would be ok with the kotlin introduction.
* Most felt that kotlin was pretty easy to do for java devs. Some were a bit skeptical that it was that similar, but open to the idea anyway.
* It was observed that the currently active server contributors were @Louka3000 @ButterscotchV @TheButlah @loucass003 @Eirenliel @Stermere. The first four were involved in the discussion and all were open or eager to the kotlin idea. After the discussion was mostly concluded and the summary posted, @Stermere and @loucass003 both said they were down to try incrementally adding support for kotlin, and that they likely preferred it over Java.
* We were worried if @Eirenliel may or may not like kotlin, and the decision ultimately will/should end up being theirs. Some asked why they would care, and it was pointed out that Eiren is still likely going to want to review all code review, so they would need to be able to do code review on kotlin and/or test kotlin code.
* **We reached a consensus that if we were to introduce kotlin, we would start with just one part of the code, or new code. This would let everyone evaluate whether they liked it or not, and if we wanted to go back to java, or keep whole parts of the codebase in java forever, that would be fine. Not everything would need to be in Kotlin, we were not prepared to make that commitment without more data.**

Eirenliel, 2022-12-16:
I am inclined to agree. Can you explain in short how the dev environment would need to be adjusted for this?

adigyran, 2022-12-16:
I have 4 years of Kotlin experience, and might help on new years weekends

adigyran, 2022-12-16:
And do code reviews  if needed

adigyran, 2022-12-16:
> I am inclined to agree. Can you explain in short how the dev environment would need to be adjusted for this?

mostly adding kotlin libraries to build.gradle, otherwise not much

Eirenliel, 2023-04-06:
We already done it.

closed 2023-04-06 by Eirenliel