SlimeVR/SlimeVR-Server #405: Upload firmware via OTA

Dounx, 2022-12-17:
Can we update every connected tracker via OTA

Dounx, 2022-12-17:
1.  Select the firmware
2.  Select the conntected trackers
3.  Click Upload button

Dounx, 2022-12-22:
I found a [firmware-update](https://github.com/SlimeVR/SlimeVR-Server/tree/firmware-update) branch,  will it implement the OTA upgrade function? @loucass003

Kamilake, 2023-01-27:
> I found a [firmware-update](https://github.com/SlimeVR/SlimeVR-Server/tree/firmware-update) branch, will it implement the OTA upgrade function?

I made this as a simple batch script.
that automatically detect the ip addr, selects and flash firmware with OTA, and then the tracker reboots itself.   

If you think this code is useful for you, I will release it with an English(or your lang)!
(currently translated into Korean)

Dounx, 2023-01-27:
> I made this as a simple batch script.
>
> that automatically detect the ip addr, selects and flash firmware with OTA, and then the tracker reboots itself.
>
>
>
> If you think this code is useful for [...]

This is exactly the functionality I want, I would like to refer to the implementation if possible, English is fine.

Kamilake, 2023-01-31:
> This is exactly the functionality I want, I would like to refer to the implementation if possible, English is fine.

If it's useful for you, I'll proceed with the work!

https://github.com/Kamilake/FineMotion_OTA_Uploader

Without python dep:
https://cdn.discordapp.com/attachments/1002402449954451466/1064140974495891476/0.3.1a_FineMotion_Updater.zip

Kamilake, 2023-01-31:
I just finished a basic translation and some tweaks! @Dounx

Dounx, 2023-01-31:
> I just finished a basic translation and some tweaks! @Dounx (It's a machine translation, but I'd appreciate it if you could review it.)

Thx, I will try to port your code to SlimeVR-Server project later.

closed 2024-12-19 by Eirenliel