SlimeVR/SlimeVR-Server #422: No feedback when triggering tap gestures.

nullstalgia, 2022-12-23:
Howdy howdy.

When using the new and (wonderful) tap reset, it's hard to tell if my taps actually went through and if I should stay in the pose for the duration of time I set for the delay, or if I should tap again, or when I can put my arms down if I have arm trackers.

A beep of some kind when the gesture is recognized, and another when the reset has occurred would be ideal.

Eirenliel, 2022-12-23:
The feature that plays a sound on reset was requested long ago, we even have a meow sound recorded for this. If someone goes to implement it, please ask me for the audio file!

marcozzxx810, 2023-02-11:
I will take this!

Louka3000, 2023-04-07:
Closed by https://github.com/SlimeVR/SlimeVR-Server/pull/576

closed 2023-04-07 by Louka3000