SlimeVR/SlimeVR-Server #446: Autobone interface improvement

Eirenliel, 2023-01-03:
* [ ] Multiple people during testing in the office didn't notice that the feet should be in one place. This should be made more clear.
* [ ] The recording time is too short, it ends while you're reading instructions. Instructions and timer should be on the same screen, with the ability to re-record it right away.

ButterscotchV, 2023-01-08:
This will hopefully be deprecated when AutoBone changes to step-based recording, though this would be a fairly simple change to improve the current implementation. *Honestly I have no idea why it was done like this, I found it super weird myself while using it and it kinda makes testing hard.*

Eirenliel, 2023-03-07:
I forgot where we discussed it, but @loucass003 is already working on what he can do with GUI. I will add new thing to this issue:

* [ ] Help users understand body proportions better by showing a stick figure with their body size on the results page. This is an advanced task, isn't required for 1.0, I think