SlimeVR/SlimeVR-Server #466: slime server stuck

zeowolfpup, 2023-01-08:
hi so the server is stuck on CONNECTING TO THE SERVER so i try to unistall it and it updated to v0.6.0 and and its got stuck on CONNECTING TO THE SERVER?

Louka3000, 2023-01-09:
Are you using [the installer](https://github.com/SlimeVR/SlimeVR-Installer/releases/latest/download/slimevr_web_installer.exe)?

Louka3000, 2023-01-09:
https://github.com/SlimeVR/SlimeVR-Installer/releases/latest/download/slimevr_web_installer.exe

medicalwei, 2023-02-26:
I found the same problem but it doesn't happen in the standalone SlimeVR build.

The following is the log running SlimeVR from command line
```
PS C:\Program Files (x86)\SlimeVR Server> java -jar .\slimevr.jar
12:09:01 [INFO] Running version v0.6.3
12:09:01 [SEVERE] java.lang.NullPointerException: Cannot invoke "com.fasterxml.jackson.databind.JsonNode.floatValue()" because the return value of "com.fasterxml.jackson.databind.node.ObjectNode.get(String)" is null
12:09:01 [SEVERE]       at dev.slimevr.config.CurrentVRConfigConverter.convert(CurrentVRConfigConverter.java:161)
12:09:01 [SEVERE]       at com.github.jonpeterson.jackson.module.versioning.VersionedModelDeserializer.deserialize(VersionedModelDeserializer.java:93)
12:09:01 [SEVERE]       at com.fasterxml.jackson.databind.deser.DefaultDeserializationContext.readRootValue(DefaultDeserializationContext.java:323)
12:09:01 [SEVERE]       at com.fasterxml.jackson.databind.ObjectMapper._readMapAndClose(ObjectMapper.java:4674)
12:09:01 [SEVERE]       at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:3666)
12:09:01 [SEVERE]       at dev.slimevr.config.ConfigManager.loadConfig(ConfigManager.java:41)
12:09:01 [SEVERE]       at dev.slimevr.VRServer.<init>(VRServer.java:80)
12:09:01 [SEVERE]       at dev.slimevr.VRServer.<init>(VRServer.java:73)
12:09:01 [SEVERE]       at dev.slimevr.Main.main(Main.kt:90)
```

medicalwei, 2023-02-26:
And the config:
```json
{"doneOnboarding":true,"debug":true,"watchNewDevices":true,"lang":"en","devSettings":{"highContrast":false,"preciseRotation":false,"fastDataFeed":false,"filterSlimesAndHMD":false,"sortByName":false,"rawSlimeRotation":false,"moreInfo":false}}
```

medicalwei, 2023-02-26:
And accompanying vrconfig.yml:
[vrconfig.yml.txt](https://github.com/SlimeVR/SlimeVR-Server/files/10832945/vrconfig.yml.txt)

Louka3000, 2023-02-27:
Thanks @medicalwei, this helps a lot ^^
Seems like the issue is with the vrconfig converter. Should be pretty easy to fix for the next release :3

Louka3000, 2023-03-08:
Fixed by https://github.com/SlimeVR/SlimeVR-Server/pull/613

closed 2023-03-08 by Louka3000