SlimeVR/SlimeVR-Server #472: Separate java codebase in workspaces

ImUrX, 2023-01-10:
So we end up having different workspaces like
- ``core``
- ``android`` (for android only stuff)
- ``desktop``
- maybe ``ios``

the main idea is that we can maintain compatibility with weird things like in android we cant use the normal serial library (even though it supports serial (insert whatever termius does here)) or ``record``s or maybe stuff like #427.

TheButlah, 2023-01-10:
You should start by defining what a workspace is. I know what it is in rust but not java/gradle

ImUrX, 2023-01-10:
It's basically different jars, you can depend between them but the idea is that we can compile an android-only jar or a desktop-only jar. [Gradle docs](https://docs.gradle.org/current/userguide/multi_project_builds.html)

Louka3000, 2023-01-10:
So if I understand correctly, we would compile several JARs and ship any of them (except `core.jar`) alongside `core.jar`, like, for example, `desktop.jar` + `core.jar`, or `android.jar` + `core.jar`.
How will this help us? 
Ideally, we could have a single .jar that works for any platform, wouldn't we? 
How would we work with different sub-projects (like what type of code would there be in the android subproject?)?
If we need something platform specific for a very small part of the code, wouldn't checking which platform the server is running on and branching out the code make more sense if that's possible?

ImUrX, 2023-01-10:
currently some changes are needed for compiling to android directly, one of those is that Android is still based on Java 8, they have some kind of transpiler from 11 to 8 so you can "kind" of use 11.
We can't also use stuff like the UnixSocket class that I added so that would *need* to be put on another workspace. 
We don't actually require to make big changes, just some medium changes on our structuring.

And how it would look would be just a ``desktop.jar`` and an ``android.jar``. We can easily include class files of ``core`` easily in gradle

ImUrX, 2023-01-10:
Btw [these are Butterscotch's efforts](https://github.com/SlimeVR/SlimeVR-Server/compare/main...ButterscotchV:SlimeVR-Server:android) to make an android jar

Louka3000, 2023-01-10:
Sounds fine to me, but I'm still wondering how working with desktop or android-specific stuff would happen with that configuration. Like, say I need to create one pipe between desktopPipe.java and androidPipe.java, how would I do it?

ImUrX, 2023-11-05:
This already happened after #727

closed 2023-11-05 by ImUrX