SlimeVR/SlimeVR-Server #502: Trackers aren't detected

Eirenliel, 2023-01-23:
On Poly's PC literally nothing happen when you plug in the tracker, even though the serial exists.

[image]
[image]
[image]
[image]

On my laptop I see a popup but it can't provision the tracker.

[image]

Even though I can actually open serial console and send GET INFO.

No idea how to even start debugging it.

Eirenliel, 2023-01-23:
After random figieting it provisioned on my laptop. I had to reboot the tracker for it to send commands, but I can't find where in the code it looks for boot sequence or anything.

0forks, 2023-01-23:
Also if the "connect trackers" window is opened for a some period of time, even just 5 minutes, it starts a lot of connect loops, resulting in many connections per second.
```
15:05:16 [INFO] Trying to open:Auto  auto: true
15:05:16 [INFO] Trying to open:Auto  auto: true
15:05:16 [INFO] Trying to open:Auto  auto: true
... 50+ lines
```

closed 2023-02-12 by Eirenliel