SlimeVR/SlimeVR-Server #509: Tap reset doesn't work

TheDevMinerTV, 2023-01-24:
In a meeting with a big person in tech we've noticed that tap reset didn't work, not sure why. I'll test around with it later.

Philip6294, 2023-01-24:
Interestingly enough, after recently reinstalling my entire PC so also reinstalling SteamVR and SlimeVR Server tap  reset also stopped working for me. 
It used to work completly fine before the reinstall

Eirenliel, 2023-01-24:
Is it possible it's disabled by default but shows as enabled? I think I had this problem with usb detection...

Philip6294, 2023-01-24:
> Is it possible it's disabled by default but shows as enabled? I think I had this problem with usb detection...

Could be, the toggles were on and had sensible delays. I didn't try to toggle them on or off tho.

Philip6294, 2023-01-29:
So after trying the latest server version: 
Quick Reset works fine but Reset and Mounting reset do not work.

Philip6294, 2023-02-03:
Some more info: 
sometimes it helped to tap on different sides of the tracker, especially the side which is facing upwards. Maybe the acceleration calculation is to blame. 
I also made a branch with logging in the TapDetectionManager and sound being played when a enough taps were registered to help with debugging this. 
https://github.com/Philip6294/SlimeVR-Server/tree/tap-detection-debug

Philip6294, 2023-02-08:
> @Philip6294 Do you mind sharing more info for your sound debugging? ( since I don't own add slime Tracker )
> 
> Can you describe the situation (any sound played or log)?
> 
>     1. when you tapped twice?
> 
>     2. when you tapped three times?

how often you tap does not matter, the sounds play after the required amount has been tapped.
After enough taps have been registered it plays the sound and after the reset it writes a line to the log.  
The logic to handle this is all added to TapDetectionManger.java

Stermere, 2023-06-17:
Do you know if this is still an ongoing issue? I believe that the cause of this is a faulty or inaccurate accelerometer. This is because to prevent false positives all other trackers must report a low acceleration for the reset to go through even if the taps are all registered. There is a config value to increase the number of trackers that can be over the threshold and still allow a reset and this has seemed to solve the problem for the few people that have used it.

Philip6294, 2023-06-18:
> Do you know if this is still an ongoing issue? I believe that the cause of this is a faulty or inaccurate accelerometer. This is because to prevent false positives all other trackers must report a [...]

I never had that problem happen to me again so I think it can be closed. 
During trying to troubleshoot it though, I noticed that low TPS could also cause taps to not register, this could maybe be added somewhere in documentation.

closed 2024-09-03 by ImUrX