SlimeVR/SlimeVR-Server #514: NoRouteToHostException

TheButlah, 2023-01-25:
```
00:44:03 [INFO] [WebSocketAPI] New connection from: 0:0:0:0:0:0:0:1
00:44:58 [INFO] [TrackerServer] Tracker timed out: udp://192.168.51.99
00:45:03 [INFO] [TrackerServer] Handshake received from /192.168.51.99:6969
00:45:03 [INFO] [TrackerServer] Sensor 0 for udp://10:91:A8:17:F4:B4 status: 1
00:45:03 [INFO] [TrackerServer] Sensor info for udp://192.168.51.99/0: 1
00:45:11 [INFO] [TrackerServer] Tracker timed out: udp://192.168.51.99
00:46:21 [SEVERE] java.net.NoRouteToHostException: No route to host
00:46:21 [SEVERE]       at java.base/sun.nio.ch.DatagramChannelImpl.send0(Native Method)
00:46:21 [SEVERE]       at java.base/sun.nio.ch.DatagramChannelImpl.sendFromNativeBuffer(DatagramChannelImpl.java:901)
00:46:21 [SEVERE]       at java.base/sun.nio.ch.DatagramChannelImpl.send(DatagramChannelImpl.java:863)
00:46:21 [SEVERE]       at java.base/sun.nio.ch.DatagramChannelImpl.send(DatagramChannelImpl.java:821)
00:46:21 [SEVERE]       at java.base/sun.nio.ch.DatagramChannelImpl.blockingSend(DatagramChannelImpl.java:853)
00:46:21 [SEVERE]       at java.base/sun.nio.ch.DatagramSocketAdaptor.send(DatagramSocketAdaptor.java:218)
00:46:21 [SEVERE]       at java.base/java.net.DatagramSocket.send(DatagramSocket.java:664)
00:46:21 [SEVERE]       at dev.slimevr.vr.trackers.udp.TrackersUDPServer.run(TrackersUDPServer.java:316)
```

TheButlah, 2023-02-07:
honestly I wasn't sure how it happened. I know it happened a lot when I was testing out my trackers. So I *think* its related to the communication with the tracker, which makes sense because this is the tracker udp server.

Most likely, we should just explicitly look for this networking error and catch it, print it, and continue on with our day, instead of crashing the thread.

ImUrX, 2023-07-26:
I know why it happens, its just when the UDP socket is trying to bind to a used port

closed 2025-09-24 by loucass003