SlimeVR/SlimeVR-Server #520: Allow user to provide SteamVR with a virtual hip tracker.

tgiddings, 2023-01-27:
Currently, the options for providing SteamVR with virtual trackers correspond to the options available in the "tracker role" setting in SteamVR for the trackers. However, some games do not use the assigned tracker role and expect the lowest torso tracker to be a hip tracker. This means that spinal rotation is interpreted as hip rotation, even if the real hips are stationary.

As a concrete example, I play VRChat and use an avatar which has tails. My SlimeVR tracking setup includes a chest tracker and a hip tracker. The "waist" tracker seen in SteamVR very accurately approximates the orientation of my real waist (which does not have an actual tracker). However, the game is not expecting a waist tracker and interprets it as a hip tracker. As such, if I turn my upper body to look behind me, my avatar's tails move even though my real hips are stationary.

This can be a cause for frustration since I naturally move my upper body to look around, but I often try to keep my avatar's hips still to avoid making distracting tail movements. I spent a long time struggling with this and only today noticed the discrepancy between which body part SlimeVR was providing tracking for and which body part VRChat expected tracking for.

Louka3000, 2023-01-27:
We do send the hip bone as the SteamVR waist tracker. The only reason it’s named this way is because SteamVR does not have a hip tracker option; only waist. Your problem is not what you think, since we are sending the hip tracker.
Make sure you’re assigning your hip tracker to hip, and not waist.

Louka3000, 2023-01-27:
Please join our Discord server for further support, since this issue’s issue doesn’t exist :p 
https://discord.gg/slimevr

closed 2023-01-27 by Louka3000