SlimeVR/SlimeVR-Server #540: Add decamove support

Louka3000, 2023-02-01:
Decamove has a sdk: https://github.com/MegadodoGames/DecaMoveSDK/

Eirenliel, 2023-02-01:
I thought they died.

Louka3000, 2023-02-01:
Idk maybe; they’re not in stock.
This is for people that bought them in the past tho. Kinects aren’t selling anymore either :p

kitlith, 2023-03-15:
This doesn't look too difficult to integrate, though managing binaries for different platforms might be interesting.

I assume we probably want to ignore positions and just get the rotations.

Should this be the domain of the server directly, or something separate akin to the wrangler for joycons? (or maybe integrated into wrangler? @carl-anders I don't know what your current plans are for expanding wrangler to handle more than joycons)

Do we know someone who actually has decamove hardware to test with or donate?

Louka3000, 2023-03-15:
Decamove is just a BNO08X. It doesn’t have position.
Someone on Discord asked for this btw.

evg-zhabotinsky, 2023-05-17:
> Decamove is just a BNO08X

Actually, they are bno0**5**5, so not even that interesting.

https://discord.com/channels/817184208525983775/823930029070876736/1091602007472742441 (decamove discussion / internals photo in #diy)

Louka3000, 2023-05-17:
Oh, that sucks.

Eirenliel, 2025-03-04:
These don't exist anymore.

closed 2025-03-04 by Eirenliel