SlimeVR/SlimeVR-Server #558: Add log file size limiter

marcozzxx810, 2023-02-05:
By @ButterscotchV, it would be quite nice to have a file size limit for the log file. Something like 50 MB or so? Anything bigger is just excessive and would waste space on people's computers.

My thoughts:
1. setup two logs (log_last_0.log, log_last_1.log) with 25MB limit
2. rotate the log when the output approximately is equal 25MB size 

Ref:
```
public FileHandler(pattern,
           int limit,
           int count)
     
Initialize a FileHandler to write to a set of files. When (approximately) the given limit has been written to one file, another file will be opened. The output will cycle through a set of count files.
The FileHandler is configured based on LogManager properties (or their default values) except that the given pattern argument is used as the filename pattern, the file limit is set to the limit argument, and the file count is set to the given count argument.

The count must be at least 1.
```
https://docs.oracle.com/javase/7/docs/api/java/util/logging/FileHandler.html#FileHandler(java.lang.String,%20int,%20int)

marcozzxx810, 2023-02-05:
@Eirenliel, any opinion about this?

Eirenliel, 2023-02-05:
sounds good

marcozzxx810, 2023-02-05:
will issue a PR for that, thanks @Eirenliel 😄

Eirenliel, 2023-02-09:
Fixed in #560.

closed 2023-02-09 by Eirenliel